XRay3D/GERBER_X3 #2: Несанкционированная фрезеровка конткра платы

Kurdl, 2019-03-27:
Импортируемый гербер-файл не содержал контура платы.
От программы не требовалось его обрабатывать, 
Но программа все равно сгенерировала Г-код с управляющей траекторией "обхода контура", рассчитав его по граничным точкам, как прямоугольник.

XRay3D, 2019-04-09:
> Но программа все равно сгенерировала Г-код с управляющей траекторией "обхода контура", рассчитав его по граничным точкам, как прямоугольник.
Так и задумывалось но данный функционал скорее всего будет перерабатываться.

Kurdl, 2019-04-09:
Мне кажется, что не следует "резать" плату автоматически.
Это для начала. 
А вообще-то Gerber-формат содержит информацию о контурах платы.
FlatCAM его распознает. Главное - не повторить ошибку FlatCAM, когда она зачем-то строит траекторию обхода этого контура: https://cnc1310.ru/Picture/GetPicture/212

closed 2019-04-11 by XRay3D