XRay3D/GERBER_X3 #3: В панели параметров инструмента цифры не вводятся с клавиатуры

Kurdl, 2019-04-10:
Не могу внести диаметр инструмента простым вводом цифр в поле "диаметр".
Сначала работают только кнопки инкремента/декремента.  После того, как 0.000 сменился на что угодно, ввод уже становится возможным.
Так же ведут себя другие элементы ввода.
Проверено на Win 7 x32  и Win 10 x64

XRay3D, 2019-04-10:
Если выделить текст (можно двойным щелчком), то поверх выделенного всё вводится.
Это вопрос к разработчикам Qt.

XRay3D, 2019-04-10:
Если выделить текст (можно двойным щелчком), то поверх выделенного всё вводится.
Это вопрос к разработчикам Qt, зачем именно так сделано.

Kurdl, 2019-04-10:
А! Так это Qt! :) Вроде ко-то говорил C++
Респект! Люблю этот тролльский продукт! ;)
А зачем для создания УП необходимо сначала выделять графическую область?
Существует возможность работать по частям?

XRay3D, 2019-04-10:
>А! Так это Qt! :) Вроде ко-то говорил C++

так он на С++ и написан. Это библиотеки, а не язык программирования.

Сейчас накидал свой контрол. Вроде работает как задумано, все стандартные заменю на свой, а то уже и самого досаждать начинает.

Kurdl, 2019-04-10:
Чтобы не приставали к автору с глупостями, может мануал какой минимальный раздать? 
Для меня, например, не все очевидно в UI

XRay3D, 2019-04-10:
Все руки ни как не дойдут, да и UI не закончен. Кстати перелазил, можете попробовать и сказать как стало, лучше или не очень.

closed 2019-04-11 by XRay3D

Kurdl, 2019-04-11:
Перезалил исходники? :) У меня компилить не чем.
По поводу контролов - не велика проблема в том, что вместо прямого ввода требуется щелкать кнопками или таскать ползунки. Мы ведь не вводим цифровое значение громкости в интерфейсе ОС.
Просто пользователь должен видеть характерный признак заблокированного для ввода контрола, например, - серый цвет.
А если еще предустановить начальные значения в рамках наиболее используемых - вообще не будет вопросов. Например, рабочий диаметр гравера - 0.1 мм. Угол - 30*  
Update: Решил кое-то проверить в программе, а она перестала открываться. Аварийно вылетает после появления на экране главного окна.

XRay3D, 2019-04-11:
> 
> 
> Перезалил исходники? :) У меня компилить не чем.

Зачем, кому мои исходники сдались? Экзешники конечно.

XRay3D, 2019-04-11:
У меня скаченное работает. А что за ошибка не пишет?

Kurdl, 2019-04-11:
Дай прямую ссылку на дистрибутив.
Я качаю с кнопки Clone/Download архив GERBER_X2-master.zip
В нем вижу только исходники.

Ошибка - стандартная Windows "Прекращена работа программы... Искать решение в интернете..."
Причем это вчерашняя версия так себя ведет. Сегодня не запускался.

XRay3D, 2019-04-11:
https://yadi.sk/d/L-laABnSVY4Ubw Качал сам у себя через браузер, всё работает.

Kurdl, 2019-04-12:
x64 работает x32 вылетает. Причем, после сноса папки и перезаписи её, где-то хранится память о последнем проекте. В реестре?

XRay3D, 2019-04-12:
Да

XRay3D, 2019-04-12:
В реестре хранится список последних файлов. В 32 разрядной версии может не хватить памяти при открытии "больших" файлов или большого количества. В будущем исправлю когда пойму как.

commit 62dc6d98 (2024-06-03): closes #3; Запомнил размер окна инструментов.
commit f0a45050 (2024-06-03): closes #3; Запомнил размер окна инструментов.