XRay3D/GERBER_X3 #4: Исправления для возможности сборки в Linux

ololoshka2871, 2019-05-07:
Попробовал собрать ваш проект под Linux.
Я использую ArchLinux.
* GCC 8.2.1
* Qt5 5.12.0-1

Я внес несколько исправлений в код для обхода ошибок компиляции.
В таком виде программа запускается и работает
[image]

Патч:
```patch
From a9faa04a242126ba4eef824fb9779358fb5b3831 Mon Sep 17 00:00:00 2001
From: ololoshka2871 <SweetTreasure@2ch.hk>
Date: Tue, 7 May 2019 08:45:01 +0300
Subject: [PATCH] =?UTF-8?q?=D0=98=D1=81=D0=BF=D1=80=D0=B0=D0=B2=D0=BB?=
 =?UTF-8?q?=D0=B5=D0=BD=D0=B8=D1=8F=20=D0=B4=D0=BB=D1=8F=20=D1=81=D0=B1?=
 =?UTF-8?q?=D0=BE=D1=80=D0=BA=D0=B8=20=D0=B2=20Linux=20(GCC=208.2=20+=20Qt?=
 =?UTF-8?q?=205.12)?=
MIME-Version: 1.0
Content-Type: text/plain; charset=UTF-8
Content-Transfer-Encoding: 8bit

---
 G2G/aboutform.cpp           | 2 ++
 G2G/filetree/fileholder.cpp | 2 +-
 G2G/gcode/toolpathcreator.h | 3 ++-
 G2G/settingsdialog.cpp      | 9 ++++++---
 G2G/settingsdialog.h        | 3 ++-
 excellon/excellondialog.cpp | 1 +
 excellon/exparser.cpp       | 1 +
 excellon/exvars.h           | 6 +++++-
 gerber/gbrfile.h            | 6 ++++--
 voronoi/halfedge.cpp        | 2 ++
 voronoi/voronoi.cpp         | 2 ++
 11 files changed, 28 insertions(+), 9 deletions(-)

diff --git a/G2G/aboutform.cpp b/G2G/aboutform.cpp
index 7f1eeab..a9a9c28 100644
--- a/G2G/aboutform.cpp
+++ b/G2G/aboutform.cpp
@@ -9,6 +9,8 @@ AboutForm::AboutForm(QWidget* parent)
     ui->setupUi(this);
 #ifdef MINGW_DDK_H
     QString str(QString(__TIMESTAMP__).append("<br/>MINGW: ") + QString::number(__MINGW32_MAJOR_VERSION) + "." + QString::number(__MINGW32_MINOR_VERSION));
+#elif __GNUG__ // specific variant for GCC
+    QString str(QString(__TIMESTAMP__).append("<br/>GCC_VER: ") + QString(__VERSION__));
 #else
     QString str(QString(__TIMESTAMP__).append("<br/>MSC_VER: ") + QString::number(_MSC_VER));
 #endif
diff --git a/G2G/filetree/fileholder.cpp b/G2G/filetree/fileholder.cpp
index ac466cb..2f1cd3a 100644
--- a/G2G/filetree/fileholder.cpp
+++ b/G2G/filetree/fileholder.cpp
@@ -1,4 +1,4 @@
-#include "FileHolder.h"
+#include "fileholder.h" // filename case-sensitive
 
 //QMap<int, QSharedPointer<Drill>> FileHolder::m_drlFiles;
 //QMap<int, QSharedPointer<G::File>> FileHolder::m_gerberFiles;
diff --git a/G2G/gcode/toolpathcreator.h b/G2G/gcode/toolpathcreator.h
index cf1858e..3fad987 100644
--- a/G2G/gcode/toolpathcreator.h
+++ b/G2G/gcode/toolpathcreator.h
@@ -31,7 +31,8 @@ void fixBegin(Path& path);
 class ToolPathCreator : public QObject {
     Q_OBJECT
 
-    friend class Clipper;
+    // friend class Clipper; // clipper.c:1619 ‘static void ToolPathCreator::progressOrCancel()’ is private within this context
+    friend class ClipperLib::Clipper;
 
 public:
     static ToolPathCreator* self;
diff --git a/G2G/settingsdialog.cpp b/G2G/settingsdialog.cpp
index 0910fa4..cae4f19 100644
--- a/G2G/settingsdialog.cpp
+++ b/G2G/settingsdialog.cpp
@@ -7,7 +7,8 @@
 
 const int gridColor = 100;
 
-const QColor defaultColor[Colors::Count]{
+//const QColor defaultColor[Colors::Count]{ // size of array ‘defaultColor’ has non-integral type ‘Colors’
+const QColor defaultColor[(size_t)Colors::Count] {
     QColor(), //Background
     QColor(255, 255, 0, 120), //Shtift
     QColor(Qt::gray), //CutArea
@@ -21,7 +22,8 @@ const QColor defaultColor[Colors::Count]{
     QColor(Qt::red) //G0
 };
 
-const QString colorName[Colors::Count]{
+//const QString colorName[Colors::Count]{ // size of array ‘colorName’ has non-integral type ‘Colors’
+const QString colorName[(size_t)Colors::Count]{
     "Background",
     "Shtift",
     "CutArea",
@@ -35,7 +37,8 @@ const QString colorName[Colors::Count]{
     "G0",
 };
 
-QColor SettingsDialog::m_color[Colors::Count]{
+//QColor SettingsDialog::m_color[Colors::Count]{ // size of array ‘SettingsDialog::m_color’ has non-integral type ‘Colors’
+QColor SettingsDialog::m_color[(size_t)Colors::Count]{
     QColor(), //Background
     QColor(255, 255, 0, 120), //Shtift
     QColor(Qt::gray), //CutArea
diff --git a/G2G/settingsdialog.h b/G2G/settingsdialog.h
index 9afb026..bf471b9 100644
--- a/G2G/settingsdialog.h
+++ b/G2G/settingsdialog.h
@@ -43,7 +43,8 @@ private:
     void onScrollBarValueChanged(int value);
     void onListCategoriesCurrentRowChanged(int currentRow);
 
-    static QColor m_color[Colors::Count];
+    //static QColor m_color[Colors::Count]; // size of array ‘m_color’ has non-integral type ‘Colors’
+    static QColor m_color[(size_t)Colors::Count];
 };
 
 #endif // SETTINGSDIALOG_H
diff --git a/excellon/excellondialog.cpp b/excellon/excellondialog.cpp
index a2c57b1..a4d516e 100644
--- a/excellon/excellondialog.cpp
+++ b/excellon/excellondialog.cpp
@@ -3,6 +3,7 @@
 #include "ui_excellondialog.h"
 #include <graphicsview.h>
 #include <scene.h>
+#include <cmath> // pow()
 
 using namespace Excellon;
 
diff --git a/excellon/exparser.cpp b/excellon/exparser.cpp
index 8ec6182..7b7bae2 100644
--- a/excellon/exparser.cpp
+++ b/excellon/exparser.cpp
@@ -2,6 +2,7 @@
 #include "exfile.h"
 
 #include <QFile>
+#include <cmath> // pow()
 
 using namespace Excellon;
 
diff --git a/excellon/exvars.h b/excellon/exvars.h
index 52b30d5..541a706 100644
--- a/excellon/exvars.h
+++ b/excellon/exvars.h
@@ -202,7 +202,11 @@ public:
     {
     }
 
-    const File* const file = nullptr;
+    // QList<T>::node_construct() -> *reinterpret_cast<T*>(n) = t; uses operator=(const Hole&),
+    // but it's deleted, because field "file" is "const",
+    // so, remove "const"
+    //const File* const file = nullptr;
+    File* file = nullptr;
     State state;
     DrillItem* item = nullptr;
 };
diff --git a/gerber/gbrfile.h b/gerber/gbrfile.h
index 0c0a86f..e9a5833 100644
--- a/gerber/gbrfile.h
+++ b/gerber/gbrfile.h
@@ -83,8 +83,10 @@ private:
         }
     }
 };
-
-Q_DECLARE_METATYPE(File)
+// specialization of ‘template<class T> struct QMetaTypeId’ in different namespace
+//Q_DECLARE_METATYPE(Gerber::File)
 }
 
+Q_DECLARE_METATYPE(Gerber::File)
+
 #endif // GFILE_H
diff --git a/voronoi/halfedge.cpp b/voronoi/halfedge.cpp
index d4ed676..3ac5164 100644
--- a/voronoi/halfedge.cpp
+++ b/voronoi/halfedge.cpp
@@ -1,5 +1,7 @@
 #include "halfedge.h"
 #include <QObject>
+#include <cmath> // std::atan2()
+
 namespace Vrn {
 Halfedge::Halfedge(Edge* edge, Vertex* lSite, Vertex* rSite)
 {
diff --git a/voronoi/voronoi.cpp b/voronoi/voronoi.cpp
index 84055b9..e0e84a1 100644
--- a/voronoi/voronoi.cpp
+++ b/voronoi/voronoi.cpp
@@ -1,6 +1,8 @@
 #include "voronoi.h"
 #include <QElapsedTimer>
 #include <QElapsedTimer>
+#include <cmath> // sqrt(), floor()
+
 namespace Vrn {
 Voronoi::Voronoi()
 {
-- 
2.20.1
```

XRay3D, 2019-05-12:
Добавил. Это нормально что GCC выдаёт столько варнингов?

closed 2019-05-12 by XRay3D

ololoshka2871, 2019-05-12:
>Это нормально что GCC выдаёт столько варнингов?
Я не собирал MSVC, так что сравнить не могу. Ну по хорошему там многовато ворнингов, можно попробовать поисправлять, Я просто не задавался такой целью, так сразу.
>Откуда узнали про прогу?
Узнал с сайта http://selenur.ru/ попробовал - понравилось, посмотрел зависимости - вроде должно работать в линуксе.

sleeply4cat, 2019-05-13:
Забавно. В пятницу начал тыкать, сломал проект, сегодня скачал заново, а оно уже готово.
Из непонятностей - у меня собралось только в Release, Debug падает с ошибками, которые, по идее, должны были сломать и релиз, однако ж.