XRay3D/GERBER_X3 #5: Снять константность с полей GraphicObject

ololoshka2871, 2019-05-14:
Структура `struct GraphicObject` используется как элемент QList, который в методе QList<GraphicObject>::node_construct() требует operator=(), однако он недоступен из-за того, что все поля константные.
Патч отменяет константность:
```patch
From 576509c1a8a02271e6a4408efaab220a63c570e1 Mon Sep 17 00:00:00 2001
From: ololoshka2871 <SweetTreasure@2ch.hk>
Date: Tue, 14 May 2019 11:03:01 +0300
Subject: [PATCH 1/2] =?UTF-8?q?gerber/gbrvars.h:=20struct=20GraphicObject?=
 =?UTF-8?q?=20=D0=A3=D0=B4=D0=B0=D0=BB=D0=B5=D0=BD=D0=B0=20=D0=BA=D0=BE?=
 =?UTF-8?q?=D0=BD=D1=81=D1=82=D1=80=D0=B0=D0=BD=D1=82=D0=BD=D0=BE=D1=81?=
 =?UTF-8?q?=D1=82=D1=8C=20=D1=81=20=D0=BF=D0=BE=D0=BB=D0=B5=D0=B9,=20?=
 =?UTF-8?q?=D1=82=D0=B0=D0=BA=20=D0=BA=D0=B0=D0=BA=20=D1=8D=D1=82=D0=BE=20?=
 =?UTF-8?q?=D0=B1=D0=BB=D0=BE=D0=BA=D0=B8=D1=80=D1=83=D0=B5=D1=82=20=D0=B3?=
 =?UTF-8?q?=D0=B5=D0=BD=D0=B5=D1=80=D0=B0=D1=86=D0=B8=D1=8E=20=D0=BE=D0=BF?=
 =?UTF-8?q?=D0=B5=D1=80=D0=B0=D1=82=D0=BE=D1=80=D0=B0=20=D0=BF=D1=80=D0=B8?=
 =?UTF-8?q?=D1=81=D0=B2=D0=B0=D0=B8=D0=B2=D0=B0=D0=BD=D0=B8=D1=8F=20=D0=BF?=
 =?UTF-8?q?=D0=BE-=D1=83=D0=BC=D0=BE=D0=BB=D1=87=D0=B0=D0=BD=D0=B8=D1=8E,?=
 =?UTF-8?q?=20=D0=BA=D0=BE=D1=82=D0=BE=D1=80=D1=8B=D0=B9=20=D1=82=D1=80?=
 =?UTF-8?q?=D0=B5=D0=B1=D1=83=D0=B5=D1=82=D1=81=D1=8F=20=D0=B4=D0=BB=D1=8F?=
 =?UTF-8?q?=20QList<GraphicObject>::node=5Fconstruct()?=
MIME-Version: 1.0
Content-Type: text/plain; charset=UTF-8
Content-Transfer-Encoding: 8bit

---
 gerber/gbrvars.h | 18 ++++++++++++------
 1 file changed, 12 insertions(+), 6 deletions(-)

diff --git a/gerber/gbrvars.h b/gerber/gbrvars.h
index 275db9b..4fefc9f 100644
--- a/gerber/gbrvars.h
+++ b/gerber/gbrvars.h
@@ -209,12 +209,17 @@ public:
     inline void setRotating(double rotating) { m_rotating = rotating; }
 };
 
+// QList<GraphicObject>::node_construct() uses operator=()
+// Gerber::GraphicObject& Gerber::GraphicObject::operator=(const Gerber::GraphicObject&)’
+// is implicitly deleted because the default definition would be ill-formed
+// so remove "const" from fields
 struct GraphicObject {
     explicit GraphicObject(
         int id,
         const State& state,
         const Paths& paths,
-        const File* gFile,
+        //const File* gFile, // invalid conversion from ‘const Gerber::File*’ to ‘Gerber::File*’
+        File* gFile,
         const Path& path = Path())
         : id(id)
         , state(state)
@@ -223,11 +228,12 @@ struct GraphicObject {
         , path(path)
     {
     }
-    const int id;
-    const State state;
-    const Paths paths;
-    const File* gFile;
-    const Path path;
+
+    int id;
+    State state;
+    Paths paths;
+    File* gFile;
+    Path path;
 };
 
 } // namespace Gerber
-- 
2.21.0
```

XRay3D, 2019-05-19:
Посмотрю. MSVC 2017 пропускает такое. Лучше наверно оператор сравнения добавлю.

ololoshka2871, 2019-05-19:
Там вот такой код
```cpp
template <typename T>
Q_INLINE_TEMPLATE void QList<T>::node_construct(Node *n, const T &t)
{
    if (QTypeInfo<T>::isLarge || QTypeInfo<T>::isStatic) n->v = new T(t);
    else if (QTypeInfo<T>::isComplex) new (n) T(t);
#if (defined(__GNUC__) || defined(__INTEL_COMPILER) || defined(__IBMCPP__)) && !defined(__OPTIMIZE__)
    // This violates pointer aliasing rules, but it is known to be safe (and silent)
    // in unoptimized GCC builds (-fno-strict-aliasing). The other compilers which
    // set the same define are assumed to be safe.
    else *reinterpret_cast<T*>(n) = t; // <--- ошибка тут
#else
    // This is always safe, but penaltizes unoptimized builds a lot.
    else ::memcpy(n, static_cast<const void *>(&t), sizeof(T));
#endif
}
```
Понятно, почему MSVC нормально проходит, он не упомянут в #ifdef
Хотя и написано, что операция безопасна, но формально синтаксически неверная.

XRay3D, 2019-05-27:
У меня всё собирается и такое даже найти не могу.

closed 2019-05-27 by XRay3D