XRay3D/GERBER_X3 #13: Исправления для сборки в linux

ololoshka2871, 2019-11-26:
Нашел время опробовать собрать новую версию.
И она собирается в Linux с минимальными исправлениями.
Вот патч.

[image]

```diff
From 810c79667b82074c4044c04b5ba9591c6cef0c25 Mon Sep 17 00:00:00 2001
From: shiloxyz <shiloxyz@q6600arch>
Date: Tue, 26 Nov 2019 17:52:03 +0300
Subject: [PATCH] Fixed linux build issues

---
 G2G/G2G.pro     | 9 ++++++++-
 G2G/project.cpp | 5 ++++-
 2 files changed, 12 insertions(+), 2 deletions(-)

diff --git a/G2G/G2G.pro b/G2G/G2G.pro
index ddd1780..f1c0408 100644
--- a/G2G/G2G.pro
+++ b/G2G/G2G.pro
@@ -21,8 +21,15 @@ msvc* {
 
 gcc* {
     QMAKE_CXXFLAGS += -std=c++1z
-    LIBS += -lsetupapi -lAdvapi32 -lpsapi
     RC_FILE = myapp.rc
+
+    win32 {
+        LIBS += -lsetupapi -lAdvapi32 -lpsapi
+    }
+}
+
+linux* {
+    DEFINES += linux
 }
 
 
diff --git a/G2G/project.cpp b/G2G/project.cpp
index c0d23d6..fc931ef 100644
--- a/G2G/project.cpp
+++ b/G2G/project.cpp
@@ -7,7 +7,10 @@
 //#include <WinNT.h>
 #include <filetree/filemodel.h>
 #include <forms/gcodepropertiesform.h>
+
+#ifndef linux
 #include <qt_windows.h>
+#endif
 
 QMap<int, QSharedPointer<AbstractFile>> Project::m_files;
 bool Project::m_isModified = false;
@@ -263,7 +266,7 @@ bool operator<(const QPair<Tool, Side>& p1, const QPair<Tool, Side>& p2)
 
 void Project::saveSelectedToolpaths()
 {
-    QVector<GCode::File*> files(files<GCode::File>());
+    QVector<GCode::File*> files(Project::files<GCode::File>());
     for (int i = 0; i < files.size(); ++i) {
         if (!files[i]->itemGroup()->isVisible())
             files.remove(i--);
-- 
2.24.0
```

XRay3D, 2019-11-27:
При условии.
/home/xr/Qt_Pro/GERBER_X2/build/G2G-Desktop_Qt_5_12_6_GCC_64bit/Debug/Getber2Gcode
[image]

closed 2019-11-27 by XRay3D

ololoshka2871, 2019-11-28:
У меня гном3, так что иконок и не видно.