XRay3D/GERBER_X3 #14: qt5 dll not found

techmike78, 2019-12-29:
I downloaded the program for x86, I start and gives an error:
[image]
My PC: hp probook 470, w10x64
Installed at https://support.microsoft.com/en-us/help/2977003.
Rebooted, but the error is the same.
I tried to install Qt5 https://www.qt.io/download-thank-you, but it also did not help, maybe something else is needed?

techmike78, 2019-12-29:
I forgot to write that such problems are only with version 0.7.8.3
For 0.7.8.2, there are a lot more files in the distribution and the program starts normally.

XRay3D, 2019-12-30:
You need to put files from 0.7.8.3 on top of 0.7.8.2.

closed 2019-12-30 by XRay3D

techmike78, 2019-12-30:
> 
> 
> You need to put files from 0.7.8.3 on top of 0.7.8.2.

This should be written on the download page or in the instructions. Beginners will definitely get confused and abandon the program.