XRay3D/GERBER_X3 #17: Добавьте поддержку linux и mac os

ah1102, 2020-01-25:
Добавьте поддержку linux и mac os

XRay3D, 2020-01-28:
> Добавьте поддержку linux и mac os

В смысле бинариков?

ah1102, 2020-01-29:
Бинарники как я заметил у вас есть. Но проект написан на Visual c.  Насколько знаю такой проект только под виндовс. У вас на форуме кто-то писал, что знает среду разработки qt. Это тот же язык c. Если код перенести в qt, то программа будет отлично компилироваться под всеми ос. Сейчас qt является наиболее востребованной. Чтобы получить исполняемый файл под linux или windows достаточно установить эту среду разработки и загрузить в нее проект.

XRay3D, 2020-02-06:
> Но проект написан на Visual c.

Можете его собрать и на маке. там как я понимаю clang и проблем с 14ым стандартом не должно быть.

ah1102, 2020-02-06:
Вы можете помочь собрать вашу программу, потому что я не особо умею это делать? Вот что я смог сделать:
```
git clone https://github.com/XRay3D/GERBER_X2.git GERBER_X2.git
$ cd GERBER_X2.git
$ mkdir build
$ cd build
$ qmake ..
$ make
```
Вот вывод в терминале:

```
$make                  
cd G2G/ && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile 
clang++ -c -pipe -stdlib=libc++ -mmacosx-version-min=10.9 -O2 -arch x86_64 -Wall -W -DQT_DEPRECATED_WARNINGS -DQT_DISABLE_DEPRECATED_BEFORE=0x060000 -DBUILD_DATE="\"?? ??? 6 16:26:47 2020\"" -DQT_DEPRECATED_WARNINGS -DQT_DISABLE_DEPRECATED_BEFORE=0x060000 -DQT_DEPRECATED_WARNINGS -DQT_DISABLE_DEPRECATED_BEFORE=0x060000 -DQT_DEPRECATED_WARNINGS -DQT_DISABLE_DEPRECATED_BEFORE=0x060000 -DQT_DEPRECATED_WARNINGS -DQT_DISABLE_DEPRECATED_BEFORE=0x060000 -DQT_DEPRECATED_WARNINGS -DQT_DISABLE_DEPRECATED_BEFORE=0x060000 -DQT_DEPRECATED_WARNINGS -DQT_DISABLE_DEPRECATED_BEFORE=0x060000 -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_GUI_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/local/Cellar/qt@4/4.8.7_6/etc/qt4/mkspecs/unsupported/macx-clang-libc++ -I../../G2G -I/usr/local/Cellar/qt@4/4.8.7_6/lib/QtCore.framework/Versions/4/Headers -I/usr/local/Cellar/qt@4/4.8.7_6/lib/QtCore.framework/Versions/4/Headers -I/usr/local/Cellar/qt@4/4.8.7_6/lib/QtGui.framework/Versions/4/Headers -I/usr/local/Cellar/qt@4/4.8.7_6/lib/QtGui.framework/Versions/4/Headers -I/usr/local/Cellar/qt@4/4.8.7_6/lib/QtOpenGL.framework/Versions/4/Headers -I/usr/local/Cellar/qt@4/4.8.7_6/lib/QtOpenGL.framework/Versions/4/Headers -I/usr/local/include -I../../clipper -I../../excellon -I../../file -I../../gcode -I../../gerber -I../../graphicsview -I/System/Library/Frameworks/OpenGL.framework/Headers -I/System/Library/Frameworks/AGL.framework/Headers/ -I. -I. -I../../G2G -I. -F/usr/local/lib -o aboutform.o ../../G2G/aboutform.cpp
In file included from ../../G2G/aboutform.cpp:1:
../../G2G/aboutform.h:17:18: warning: 'override' keyword is a C++11 extension [-Wc++11-extensions]
    ~AboutForm() override;
                 ^
../../G2G/aboutform.cpp:19:40: error: 'clicked' is a protected member of 'QAbstractButton'
    connect(ui->cmdOk_2, &QPushButton::clicked, this, &AboutForm::accept);
                                       ^
/usr/local/include/QtGui/qabstractbutton.h:127:10: note: declared protected here
    void clicked(bool checked = false);
         ^
../../G2G/aboutform.cpp:19:5: error: no matching member function for call to 'connect'
    connect(ui->cmdOk_2, &QPushButton::clicked, this, &AboutForm::accept);
    ^~~~~~~
/usr/local/include/QtCore/qobject.h:204:17: note: candidate function not viable: no known conversion from
      'void (QAbstractButton::*)(bool)' to 'const char *' for 2nd argument
    static bool connect(const QObject *sender, const char *signal,
                ^
/usr/local/include/QtCore/qobject.h:217:17: note: candidate function not viable: no known conversion from
      'void (QAbstractButton::*)(bool)' to 'const QMetaMethod' for 2nd argument
    static bool connect(const QObject *sender, const QMetaMethod &signal,
                ^
/usr/local/include/QtCore/qobject.h:337:22: note: candidate function not viable: no known conversion from
      'void (QAbstractButton::*)(bool)' to 'const char *' for 2nd argument
inline bool QObject::connect(const QObject *asender, const char *asignal,
                     ^
../../G2G/aboutform.cpp:20:36: error: 'linkActivated' is a protected member of 'QLabel'
    connect(ui->lblAbout, &QLabel::linkActivated, [](const QString& link) { QDesktopServices::openUrl(link); });
                                   ^
/usr/local/Cellar/qt@4/4.8.7_6/lib/QtGui.framework/Versions/4/Headers/qlabel.h:135:10: note: declared protected here
    void linkActivated(const QString& link);
         ^
../../G2G/aboutform.cpp:20:51: error: expected expression
    connect(ui->lblAbout, &QLabel::linkActivated, [](const QString& link) { QDesktopServices::openUrl(link); });
                                                  ^
1 warning and 4 errors generated.
make[1]: *** [aboutform.o] Error 1
make: *** [sub-G2G-make_default] Error 2
```
Я подумал, что может ошибки из-за qt4 и установил qt5:

```
$brew install qt5
$brew unlink qt@4
$brew link qt5 --force
Linking /usr/local/Cellar/qt/5.14.1... 497 symlinks created
$qmake ..
$make

fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[1]: *** [abstractnode.o] Error 1
make: *** [sub-G2G-make_first] Error 2
```
[qmake qt5.txt](https://github.com/XRay3D/GERBER_X2/files/4165636/qmake.qt5.txt)

ololoshka2871, 2020-02-11:
Вероятно, причина в следующем:  `-std=c++1z -O2 -std=gnu++11` А должно быть `-std=c++14`
Мака не имею, поэтому проверить не могу.

OllyCat, 2020-03-03:
Добрый день.

Не подскажите в чём проблема? Попытка сборки под arch linux, qt 5.14, gcc 9.2.1.
Ругается на в куче разных мест на

error: ambiguous overload for ‘operator<<’ (operand types are ‘QDataStream’ and ‘const Excellon::ZeroMode’)
  150 |         stream << fmt.zeroMode;
      |         ~~~~~~ ^~ ~~~~~~~~~~~~
      |         |             |
      |         QDataStream   const Excellon::ZeroMode

Кусок вывода:

`cd G2G/ && ( test -e Makefile || /usr/bin/qmake -o Makefile /home/olly/work/AUR/gerber_x2/src/GERBER_X2/G2G/G2G.pro ) && make -f Makefile 
make[1]: Entering directory '/home/olly/work/AUR/gerber_x2/src/GERBER_X2/build/G2G'
g++ -c -pipe -std=c++1z -O2 -Wall -Wextra -D_REENTRANT -fPIC -DQT_DEPRECATED_WARNINGS -DQT_DISABLE_DEPRECATED_BEFORE=0x060000 -Dlinux -DBUILD_DATE="\"Tue Mar 3 11:28:10 2020\"" -DQT_DEPRECATED_WARNINGS -DQT_DISABLE_DEPRECATED_BEFORE=0x060000 -DQT_DEPRECATED_WARNINGS -DQT_DISABLE_DEPRECATED_BEFORE=0x060000 -DQT_DEPRECATED_WARNINGS -DQT_DISABLE_DEPRECATED_BEFORE=0x060000 -DQT_DEPRECATED_WARNINGS -DQT_DISABLE_DEPRECATED_BEFORE=0x060000 -DQT_DEPRECATED_WARNINGS -DQT_DISABLE_DEPRECATED_BEFORE=0x060000 -DQT_DEPRECATED_WARNINGS -DQT_DISABLE_DEPRECATED_BEFORE=0x060000 -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_PRINTSUPPORT_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CONCURRENT_LIB -DQT_CORE_LIB -I../../G2G -I. -I../../clipper -I../../excellon -I../../file -I../../gcode -I../../gerber -I../../graphicsview -isystem /usr/include/qt -isystem /usr/include/qt/QtOpenGL -isystem /usr/include/qt/QtPrintSupport -isystem /usr/include/qt/QtWidgets -isystem /usr/include/qt/QtGui -isystem /usr/include/qt/QtConcurrent -isystem /usr/include/qt/QtCore -I. -I. -I/usr/lib/qt/mkspecs/linux-g++ -o abstractnode.o ../../G2G/filetree/abstractnode.cpp
In file included from ../../excellon/exfile.h:4,
                 from ../../G2G/project.h:8,
                 from ../../G2G/filetree/abstractnode.cpp:3:
../../excellon/extypes.h: In function ‘QDataStream& Excellon::operator<<(QDataStream&, const Excellon::Format&)’:
../../excellon/extypes.h:150:16: error: ambiguous overload for ‘operator<<’ (operand types are ‘QDataStream’ and ‘const Excellon::ZeroMode’)
  150 |         stream << fmt.zeroMode;
      |         ~~~~~~ ^~ ~~~~~~~~~~~~
      |         |             |
      |         QDataStream   const Excellon::ZeroMode
In file included from /usr/include/qt/QtGui/qregion.h:49,
                 from /usr/include/qt/QtGui/qmatrix.h:45,
                 from /usr/include/qt/QtGui/qpainterpath.h:44,
                 from /usr/include/qt/QtWidgets/qgraphicsitem.h:48,
                 from /usr/include/qt/QtWidgets/QGraphicsItem:1,
                 from ../../graphicsview/scene.h:4,
                 from ../../G2G/point.h:4,
                 from ../../G2G/mainwindow.h:4,
                 from ../../G2G/filetree/abstractnode.cpp:2:
/usr/include/qt/QtCore/qdatastream.h:178:18: note: candidate: ‘QDataStream& QDataStream::operator<<(qint8)’
  178 |     QDataStream &operator<<(qint8 i);
      |                  ^~~~~~~~
<skip>...`

ololoshka2871, 2020-03-11:
@OllyCat 
Это все из-за Qt 5.14, попорбуйте с патчем #19 
Заодно протестируите.

ah1102, 2020-03-11:
> Вероятно, причина в следующем: `-std=c++1z -O2 -std=gnu++11` А должно быть просто `-std=c++1z`
> Мака не имею, поэтому проверить не могу.

Простите за глупый вопрос, но как применить эти изменения, а вернее, что и куда нужно дописать?

OllyCat, 2020-03-19:
> @OllyCat
> Это все из-за Qt 5.14, попорбуйте с патчем #19
> Заодно протестируите.

Был в отъезде, только сейчас руки дошли. Всё собралось, запустилось с этим патчем. Спасибо большое. Буду тестировать.

ah1102, 2020-04-10:
Опытные товарищи могут подсказать по моему вопросу? Не понимаю в чем сложность просто подсказать. Я пробовал гуглить, но ответа не нашел.

ah1102, 2020-04-13:
Проект заглох? 3-месяца назад последнее обновление и никто не может ответить.

XRay3D, 2020-05-09:
> Проект заглох? 3-месяца назад последнее обновление и никто не может ответить.

linux собирается, а mac-os ответит тот кто на ней работает.

ah1102, 2020-05-09:
Пока ждал ответа научился собирать flatcam https://youtu.be/eQaK3m9j8rs Очень редко отвечают в ветке вашего приложения, к сожалению((( 

А вы как опытный специалист, можете мне сказать, вот эти настройки под комментарием, они указываются в параметрах командной строки или как внесение изменений в исходные файлы проекта?

> Вероятно, причина в следующем: -std=c++1z -O2 -std=gnu++11 А должно быть просто -std=c++1z
> Мака не имею, поэтому проверить не могу.

XRay3D, 2020-05-09:
Это флаги компилятора для выбора стандарта языка.

anal-probe, 2021-03-12:
[image]
коллеги, оживлю тред -- пытаюсь вот собрать на маке, ошибку выдает. это что, не та версия qt?

XRay3D, 2021-03-14:
Linux In 0.8.0 Resolved

XRay3D, 2021-03-14:
> [image]
> коллеги, оживлю тред -- пытаюсь вот собрать на маке, ошибку выдает. это что, не та версия qt?
Нужен компилер поддерживающий С++20, у меня нет мака чтобы разобраться с этим.

djbios, 2021-03-20:
Долго мучался со сборкой, в итоге собрал для Linux Mint 19, вдруг кому пригодится
[GGEasy.tar.gz](https://github.com/XRay3D/GERBER_X3/files/6175410/GGEasy.tar.gz)

XRay3D, 2021-03-21:
> Долго мучался со сборкой, в итоге собрал для Linux Mint 19, вдруг кому пригодится
> [GGEasy.tar.gz](https://github.com/XRay3D/GERBER_X3/files/6175410/GGEasy.tar.gz)

Проблемы в коде были или в смаке? Я собрать то собрал, но не понял как правильно всё собрать и упаковать для шары. Гляну вашу версию на предмет запуска у меня в кедах.

anal-probe, 2021-03-22:
на маке собралось, тоже пришлось помучаться, ибо далек я от с++ :)
вроде работает. пока нашел только одну ошибку - вылетает при нажатии "закрыть" настроек сверловки

[image]

в консоли segmentation fault  /Temp/GERBER_X3/bin/GGEasy_gnu_x64_d

если закрывать панель "крестиком" такой проблемы нет. пока удалил эту кнопку с формы, чтобы не смущала)

djbios, 2021-03-22:
> Проблемы в коде были или в смаке? Я собрать то собрал, но не понял как правильно всё собрать и упаковать для шары. Гляну вашу версию на предмет запуска у меня в кедах.

Cmake. Сбилдилось только со статичной линковкой, изменения в ПР #48

anal-probe, 2021-03-22:
> если закрывать панель "крестиком" такой проблемы нет. пока удалил эту кнопку с формы, чтобы не смущала)
эта проблема все-таки присутствует: если оставить открытую панельку, например вороной, и не закрывая открыть панельку со сверловкой -- вылетает. если предварительно закрыть панельку крестиком -- все ок. 
можно как-то поправить?

XRay3D, 2021-03-28:
> эта проблема все-таки присутствует: если оставить открытую панельку, например вороной, и не закрывая открыть панельку со сверловкой -- вылетает. если предварительно закрыть панельку крестиком -- все ок.

Сейчас проверил и не нашёл проблем, попробуйте мастера стянуть и собрать.

anal-probe, 2021-03-28:
неа, проблема на месте. плюс сейчас добавилось, что при запуске программа встречает сразу диалогом
[image]

и не сохраняет Exellon настройки при нажатии ОК

GalaktionovPankrat, 2021-08-02:
Господа, которые смогли установить данную программу на Linux, напишите пожалуйста подробную инструкцию о том, как Вы это сделали? Что еще нужно устанавливать и где это брать. С Linux работать не умею, но вынужден пользоваться в силу необходимости.

Дистрибутив Debian, если это важно. Заранее благодарю того, кто осмелится на этот труд.

XRay3D, 2021-11-20:
> неа, проблема на месте. плюс сейчас добавилось, что при запуске программа встречает сразу диалогом [image]
> 
> и не сохраняет Exellon настройки при нажатии ОК

Окно это запуск тестов в конструкторе майнвиндов, а настройки сверловок пофикшены в 0.8.0.3

ah1102, 2022-04-09:
> на маке собралось, тоже пришлось помучаться, ибо далек я от с++ :) вроде работает. пока нашел только одну ошибку - вылетает при нажатии "закрыть" настроек сверловки
>
> [image]
>
> в консоли segmentation [...]

Как собрать? Поделитесь!

ronindev, 2023-04-15:
Хотелось бы инструкцию для сборки под Linux.

kpbicka, 2026-02-26:
> [@kpbicka](https://github.com/kpbicka) Вам удалось получить работающий GGeasy на Mac?

@schadling я давно уже работаю в более продвинутом FlatCam - он отлично работает на  Mac