XRay3D/GERBER_X3 #25: Баг при разбиение Вороного.

iMaks-RS, 2020-06-25:
Добрый день!

Опять вылез баг при разбиение Вороного.. один пад остался висеть в воздухе, отрезанный от все остальных. Проект не сохранил, т.к. спешил. ((( Использовал тестую версию G2G. Выгружал из диптрейс в весрию Gerber X2

XRay3D, 2020-06-30:
Файлик бы не помешал, похоже диптрейс как то особенно генерит гербер и я его некорректно разбираю.

iMaks-RS, 2020-08-04:
При импорте "потерялась" одна дорожка.. По всей видимости это и есть причина.. Потеря дорожки при импорте в G2G.. если этот момент не проверить.. то потом на плате приходится кидать перемычки (((
[image]

версии используемого ПО.  
[image]
[image]

если повторно импортировать файл - бага нет и дорожка появляется

[lan adapter.zip](https://github.com/XRay3D/GERBER_X2/files/5021469/lan.adapter.zip)

lskramarov, 2020-08-04:
при импорте герберов сгенеренных в альтиуме тоже иногда, что то не прогружается, так же помогает повторный импорт.

iMaks-RS, 2020-08-06:
> 
> 
> при импорте герберов сгенеренных в альтиуме тоже иногда, что то не прогружается, так же помогает повторный импорт.

Поставил самую актуальную на данный момент версию G2G, не тестовую, 784.  Проблема с пропажей дорожки при импорте ушла.  На версии 778 этот баг всегда появляется при первом же импорте гербера. Иногда приходилось раза по 3-4 открывать файл, пока не откроет без глюк ))) Альтиум не владею, но хочется.. т.к. модуль встроен во Fusion360..  Тогда всю подготовку к фрезеровке можно будет в одном софте делать )

closed 2021-01-21 by XRay3D