XRay3D/GERBER_X3 #30: Сделать нарезку ненужной меди на квадратики как альтернативу полной выборки

El-Engr, 2021-02-09:
Свободная незаземленная медь между дорожками может стать причиной паразитных связей. Для ВЧ плат ее просто необходимо убирать, хотя это требует времени и ресурса гравера. Но проблему таких паразитных связей можно решить, нарезав медь на мелкие квадратики наподобие "Copper Thieving". Это значительно ускорит  изготовление платы и сохранит ресурс гравера. В принципе, достаточно разрешить устанавливать отступ больше, чем половина диаметра гравера, но специальный режим был бы весьма удобен.

XRay3D, 2021-02-15:
Я подумаю над этим. Как это правильно обозвать.

El-Engr, 2021-02-15:
Спасибо!
По-английски, наверное, правильно будет "Crosshatching" или "Copper crosshatching", ну или что-то подобное. Термин "Copper thieving" употреблять, думаю, не стоит, это делается для другой цели. По-русски выходит "Перекрестная насечка" или "Насечка свободной меди" и т.п.

XRay3D, 2021-02-21:
[image]
Типа такого?

closed 2021-02-21 by XRay3D

El-Engr, 2021-02-21:
Да! Отлично!