XRay3D/GERBER_X3 #31: Закрывается при попытке сохранения УП сверловки

PNechNET, 2021-02-11:
При сохранении УП сверловки закрывается. Ошибок не выдает. Сверловка в формате .xln

[213.zip](https://github.com/XRay3D/GERBER_X3/files/5964952/213.zip)

kfmut, 2021-02-15:
@PNechNET программка вроде бы падает, если для сохранения выбирать УП, кликая ПКМ на колонках "Сторона" или "Тип", тогда имя файла вместо нормального отображается в виде "Верх"/"Низ" или остаётся пустым. После жмакания по "Сохранить", падает.  Если жмакать по колонке "Название", то всё ОК. У вас также или по-другому?

PNechNET, 2021-02-15:
На данный момент перешел на, пока что, бета версию. Найти ее можете на форуме [(https://www.cnc-club.ru/forum/viewtopic.php?f=155&p=594920#p594920)]. По этому, пока что не закрываю тему.

PNechNET, 2021-02-15:
> @PNechNET программка вроде бы падает, если для сохранения выбирать УП, кликая ПКМ на колонках "Сторона" или "Тип", тогда имя файла вместо нормального отображается в виде "Верх"/"Низ" или остаётся [...]

Ответил выше

XRay3D, 2021-02-15:
> 
> 
> На данный момент перешел на, пока что, бета версию. Найти ее можете на форуме https://www.cnc-club.ru/forum/viewtopic.php?f=155&p=594920#p594920. По этому, пока что не закрываю тему.
В 0.8.0 это и многое другое поправлено.

XRay3D, 2021-02-23:
Fixed in 0.8.0

closed 2021-02-23 by XRay3D