XRay3D/GERBER_X3 #32: Предложения https://www.cnc-club.ru/forum/viewtopic.php?p=601930#p601930

XRay3D, 2021-02-15:
1. Софтина вылетает при попытке сохранить файл УП. Любой. Сейчас проверю конечно на свежих бетах, но однозначно воспроизводится на конфигурации 0.7.9.0 (build 28/12/2020), Win10 (кстати, нельзя скопировать версию из окошка "O G2G"). Генерит УП нормально - т.е. приходится "показать источник" на УП и копировать текст ручками...
2. Хотелось бы в "настройки" - "G-код" возможности привязать "формат строки с координатами" к машине (фрезер/лазер). В частности различия вызывает S-параметр с мощностью - для лазера он актуален (а-ля HLDI), для фрезера только мешается.
3. Хочется уметь задавать позиции ноля координат относительно нижней левой булавки (сейчас задаётся относительно нижнего левого угла полезного содержимого платы). Зачем: на жертвенном столе имеется отверстие под эту булавку с заведомо известными координатами. Булавки же в зависимости от характера платы/зазоров до границ текстолита могут плавать (ну например, плата по X относительно заготовки существенно меньше - я уношу булавки в X, всё равно обрезки останутся. напротив, плата в притык к границам текстолита - булавки уезжают в Y). Соответственно сейчас это пересчитывается и переставляется 0 УП руками, но бывает забываю/ошибаюсь да и в целом кажется что возможность задавать 0 относительно реальных границ реза (а не только границ полезного содержимого) должна быть полезна многим.
4. Возможность программного управления - процесс изготовления довольно типовой и хотелось бы (скриптами дёргая софтину с командной строки или ещё как) иметь возможность "в один клик" получать набор УП для разных этапов. Да, понимаю, что не для всех, но у меня например процесс очень типовой и даже файлы называю однотипно

Код: Выделить всё

0.1-Сверловка реперов
0.2-Отрез заготовки
1-Засветка фоторезиста
2-Засветка маски
3.x-Сверление/фрезеровка отверстий (по количеству рабочих инструментов)
4-Отрезка платы

Соответственно, единственное ручное тут - это добавление линии для отреза куска платы от остальной заготовки, остальное - однотипное накликивание.
5. Собственно, возможность обозначать как-то на рабочем поле саму изначальную заготовку (прямоугольный кусок текстолита заведомо известного размера) с возможностью подвигать относительно платы и автоматом сгенерить линии отрезки. Сейчас обхожусь фигурой "прямоугольник", но во-первых задавать его размеры через точки углов совсем не удобно, во-вторых хотелось бы фич авто-позиционирования платы (в идеале - включая поворот, а то я повернуть плату умею только перегенерируя gerber) и авто-генерации линий отреза маленькой заготовки под плату от большой.

PS: Мелочи:
6. Хотелось бы возможность удалить конкретную булавку. Как уже писал выше - я вырезаю из g-кода руками сверловку первой булавки - отверстие под неё уже есть на столе, а на заготовке сверлю ручками.
7. При перемещении булавок мышкой на поле - игнорируется привязка к сетке.
8. Почему-то при установке в "параметры g-кода" значений ноля координат выставляю, скажем, -10 - а после сохранения/открытия - там всё равно -10,010 или -10,025. Видимо, какая-то дискретность и странное округление.

LinFor, 2021-02-25:
Привет, раскидал по разным issues актуальное.

XRay3D, 2021-03-02:
Хорошо. Тогда закрываю этот.

closed 2021-03-02 by XRay3D