XRay3D/GERBER_X3 #37: 0.8.0 preview (GGEasy_x32) и beta 0.8.0

YuryW, 2021-02-22:
Выделенные элементы ПП с помощью Ctrl и мышки по цвету не отличаются от ещё не выделенных , что затрудняет этот этап работы.

XRay3D, 2021-02-23:
> Выделенные элементы ПП с помощью Ctrl и мышки по цвету не отличаются от ещё не выделенных , что затрудняет этот этап работы.
А можно по подробнее? У меня всё работает, можно скриншот.

YuryW, 2021-02-23:
Работает, только когда наводишь мышкой на выбранные компоненты, то есть можно посмотреть какие выбрал, обязательно наведя на них курсор. Как только курсор съезжает с элемента, цвет у всех становится одинаковым.

YuryW, 2021-02-23:
Решение проблемы было в настройках видеокарты.
Цифровую интенсивность уменьшил со 100% до 50% и всё стало четко просматриваться.
Спасибо!

XRay3D, 2021-02-23:
Никогда бы не догадался туда залезть.

closed 2021-02-23 by XRay3D