XRay3D/GERBER_X3 #40: Разметка заготовки текстолита

LinFor, 2021-02-25:
Возможность обозначать как-то на рабочем поле саму изначальную заготовку (прямоугольный кусок текстолита заведомо известного размера) с возможностью подвигать относительно платы и автоматом сгенерить линии отрезки. Сейчас обхожусь фигурой "прямоугольник", но во-первых задавать его размеры через точки углов совсем не удобно, во-вторых хотелось бы фич авто-позиционирования платы (в идеале - включая поворот, а то я повернуть плату умею только перегенерируя gerber) и авто-генерации линий отреза маленькой заготовки под плату от большой.

Sacha71, 2021-12-17:
Поддерживаю поворот платы было бы здорово. Но если это трудно реализуется то лучше программу всякими возможностями не перегружать. Главное что бы то что есть работало надёжно как молоток. :)

atmelletje, 2022-01-04:
Very well done, it's easy to use and resonable fast generating the result. I used several different Cad programs and all give the same good result. I have an auto tool change CNC machine (Home Made). Is it possible to generate all the needed steps in one file?. I haven't found this option till now.

XRay3D, 2022-09-16:
> Very well done, it's easy to use and resonable fast generating the result. I used several different Cad programs and all give the same good result. I have an auto tool change CNC machine (Home Made). Is it possible to generate all the needed steps in one file?. I haven't found this option till now.

I'll think about it.

XRay3D, 2022-09-16:
> Возможность обозначать как-то на рабочем поле саму изначальную заготовку (прямоугольный кусок текстолита заведомо известного размера) с возможностью подвигать относительно платы и автоматом сгенерить линии отрезки. Сейчас обхожусь фигурой "прямоугольник", но во-первых задавать его размеры через точки углов совсем не удобно, во-вторых хотелось бы фич авто-позиционирования платы (в идеале - включая поворот, а то я повернуть плату умею только перегенерируя gerber) и авто-генерации линий отреза маленькой заготовки под плату от большой.
> 
> From #32

Такой вариант устроит [https://disk.yandex.ru/d/L-laABnSVY4Ubw](url)? В крайней версии на файлах выделенных правовой кнопкой в меню выбрать трансформация и двигать вращать как угодно

closed 2022-09-23 by XRay3D