XRay3D/GERBER_X3 #46: Перенести "ускорение" в свойства инструмента

LinFor, 2021-03-15:
На данный момент параметр "ускорение" задаётся в свойствах конкретной процедуры обработки.
При этом фактически, ускорение ограничено самим рабочим инструментом (его весом) и жёсткостью механики станка.
Предлагаю:
1. Перенести настройку ускорения в свойства рабочего инструмента - т.е. туда же, где находятся остальные скоростные характеристики.
2. В стартовый/финишный g-code добавить макроподстановку со значением ускорения для текущей УП - для использования M204/M201

XRay3D, 2021-03-15:
>     2. В стартовый/финишный g-code добавить макроподстановку со значением ускорения для текущей УП - для использования M204/M201

А вот тут по подробнее.

LinFor, 2021-03-15:
> > ```
> > 2. В стартовый/финишный g-code добавить макроподстановку со значением ускорения для текущей УП - для использования M204/M201
> > ```
> 
> А вот тут по подробнее.

Ну эмм. Возможность указать в стартовом g-code не просто "M204 P250 T250", но и что-нить вроде "M204 P{tool_acceleration} T{tool_acceleration}", при этом сам tool_acceleration берётся из свойств рабочего инструмента, а в итоговый генерируемый программой g-code попадает всё равно "M204 P250 T250".

Возможно нужно пойти вообще другим путём - занести в свойства инструмента возможность указать куски g-code, вставляемые при разных событиях инструмента. Только затрудняюсь сразу предположить набор этих событий. Ну для того же лазера это инициализация (переключение на инструмент), включение и выключение луча. Для фрезы сложнее, т.к. начало/конец реза - это движения по оси Z, а не просто "M3".

XRay3D, 2021-03-19:
В GRBL нету такого. Да и ускорение вводится один раз в окне растра, и больше не трогается. Так как это "свойство" станка а не инструмента.

LinFor, 2021-03-19:
Да судя по документации всё то же самое в GRBL, только синтаксис другой:
```
$120 = 10.000    (X-axis acceleration, mm/sec^2)
$121 = 10.000    (Y-axis acceleration, mm/sec^2)
$122 = 10.000    (Z-axis acceleration, mm/sec^2)
```

Насчёт свойства станка - и да и нет. Если у меня на одном станке на одной каретке будут кататься то тяжёлый шпиндель, то лёгкий лазер - то это вполне себе свойство инструмента.
Были б отдельно профили станков - да, было б логичней туда.

Насчёт не трогается - я в зависимости от операции и требуемой точности (маска, резист) свечу с разной скоростью и ускорениями, т.к. на больших ускорениях звон появляется - иногда на это пофиг, а иногда уже и нет. Плюс маска более мощной головой засвечивается. Соответственно заведено это всё разными инструментами.
Шаблонов операций тоже хотелось бы, либо программного управления - ручками путается, какой слой засвечивать "внутри", какой "снаружи", ну и вот это вот всё :)

YuryW, 2021-03-25:
$120
$121
$122
Это настройки GRBL, или если быть точными по данной теме, то настройки конкретного "железа"  или станка. Так что ускорение, на мой взгляд, должно остаться индивидуально в настройках HLDI, где оно и используется GGEasy для создания g-кода. 
А оператор должен не забывать установить в настройках GRBL ускорение, не меньшее, чем в GGEasy. И только тогда всё получится.

closed 2021-03-26 by LinFor