XRay3D/GERBER_X3 #47: Ошибки в редактировании базы инструментов

kfmut, 2021-03-15:
Добрый!

Для релиза 0.8.0 заново заполнял базу инструментов, заметил несколько ошибок:
- в параметрах резания "Отступ" в численном выражении можно теперь крутить до диаметра инструмента, а проценты так и остаются на 50%;
- если в панельке "Информация об инструменте" стоит чек-бокс "Скорость погружения", и сделать переход с инструмента типа "Сверло" на инструмент типа "Фреза", то скорость погружения для инструмента считается по "Резу за оборот", а не берётся та, которая была задана при создании инструмента. При переходе между "Фрезой" и "Фрезой" всё нормально;
- проверку корректности вводимого значения для "Реза за оборот" оооочень бы хотелось видеть какой-то более простой чем сейчас, т.к. невозможно забить значение просто введением на цифровой части клавиатуры последовательности цифр/точки, типа "0,025".

kfmut, 2021-03-15:
В первом пункте не совсем правильно сформулировал:
- проверка на "Оступе" в численном значении отваливается, если в базе инструментов перейти с инструмента с бОльшим диаметром на инструмент с меньшим диаметром, вот тогда отступ можно крутить колёсиком мышки свыше половины диаметра. И всплывающая подсказка тогда же говорит, что "Отступ" может быть до диаметра.

Баг/не баг, но вообще хотелось бы иметь возможность "Отступ" задавать до полного диаметра инструмента штатно :-)

Bob-eremeev, 2021-08-08:
Всё же это баг. Так как, значения отступа в абсолютных и относительных единицах перестают соответствовать друг другу. Начинает он проявляться при переходе от инструмента с меньшим диаметром к инструменту с большим, и сохраняется вдальнейшем при любых переходах. Кроме того, изменять отступ в абсолютных значениях можно и стрелками, а не только колёсиком мышки.

kfmut, 2021-08-08:
@Bob-eremeev

В крайней разработческой версии проверку эту поменяли, можно крутить отступ больше 50%, но получилось с особенностями:
https://www.cnc-club.ru/forum/viewtopic.php?p=615935#p615935