XRay3D/GERBER_X3 #54: Не читает dxf

kirr2, 2021-07-22:
Подскажите, как и при каких условиях (пример программ, через которые делался экспорт dxf) читается dxf?
Пробовал через корел, разные версии годов, не видит контуров никаких.
Спасибо

YuryW, 2021-07-22:
Make Box 2.7
https://www.cnc-club.ru/forum/viewtopic.php?p=614271#p614271
Там же на форуме в теме можно найти ссылку на эту программу....

kirr2, 2021-07-23:
Ссылку на программу не нашел, в поисковике не котируется тоже. Если не трудно, можно ссылку, 88 страниц форума не найду, на первой странице тоже никаких ссылок...

YuryW, 2021-07-24:
Первоисточник:
https://www.radiokot.ru/forum/viewtopic.php?f=9&t=135432&start=60
(Прокрутить всю страницу вниз, предпоследний пост пока...)

YuryW, 2021-07-24:
Возможно, тоже пригодится (в нагрузку):
https://cnc3018.ru/threads/ggeasy-grber2gcode-frezerovka-pechatnyx-plat-na-chpu-zasvetka-lazerom-izuchaem-delimsja-opytom-i-dostizhenijami.2231/post-86031

XRay3D, 2021-07-24:
> Подскажите, как и при каких условиях (пример программ, через которые делался экспорт dxf) читается dxf?
> Пробовал через корел, разные версии годов, не видит контуров никаких.
> Спасибо

А есть файлы для "поглазеть"?

kirr2, 2021-07-25:
Да, можете скинуть почту, перешлю файлы

kirr2, 2021-07-25:
Нашел, переслал файлы и небольшие комментарии, надеюсь все получится)

YuryW, 2021-07-25:
> А есть файлы для "поглазеть"?

Программа LibreCAD (версия 2.2.0), нарисованы несколько фигур для теста и сохранены затем в пяти видах dxf - 2007, 2004, 2000, R14 и R12... GGEasy открыла их все, но...
Эллипс как одна из исходных нарисованных фигур отобразился только в версии DXF R12...
[0.8.0.2_dxf - копия.zip](https://github.com/XRay3D/GERBER_X3/files/6873798/0.8.0.2_dxf.-.zip)

YuryW, 2021-07-26:
Похоже, в моём случае вопрос в поддержке многочленов (полиномов), кусочнополиноминальная аппроксимация. 
LibreCAD, линия сплайн, сохранена в пяти DXF - всё тех же. GGEasy открыла только DXF R12, А формат R12 (при сохранении в нём) все эти формулы подменяет в отрезки, дуги и круги: 
"...При сохранении рисунка в формате Release 12/LT 2 DXF......
Компактные полилинии и образцы штриховки преобразуются в полилинии и образцы штриховки формата R12.
Все тела, ACIS-тела, области, эллипсы, выноски, мультилинии, лучи, рамки допусков и прямые преобразуются в отрезки, дуги и круги...."
[0.8.0.2_dxf_сплайн - копия.zip](https://github.com/XRay3D/GERBER_X3/files/6876950/0.8.0.2_dxf_.-.zip)

kirr2, 2021-07-27:
Сохранил из CorelDraw (точнее экспортировал) в формате dxf R11 (R12 нету) - прочитал. R13 - не читает.

Тем не менее, меня этот результат уже устраивает, буду сохранять в формате AutoCAD R11.
Не нашел, правда, как двигать контуры по рабочему полю или переназначать нулевую (origin) точку начала отсчета, например, привязаться к нижнему левому углу прямоугольника. Попробую потыкаться.

YuryW, 2021-07-27:
> Не нашел, правда, как двигать контуры по рабочему полю...

Левой кнопкой мыши выделяете нужные фигуры чертежа, нажав и удерживая Ctrl. После этого кликаете Авторазмещение всех опорных точек.

> Не нашел, правда, как ..... переназначать нулевую (origin) точку начала отсчета, например, привязаться к нижнему левому углу прямоугольника. .....

Инструменты -> Параметры -> (Настройки) Утилиты -> Нуль УП -> Место

XRay3D, 2021-07-27:
> Сохранил из CorelDraw (точнее экспортировал) в формате dxf R11 (R12 нету) - прочитал. R13 - не читает.
>
> Тем не менее, меня этот результат уже устраивает, буду сохранять в формате AutoCAD R11.
> Не [...]

В итоге всё-таки открывает?

kirr2, 2021-07-28:
> В итоге всё-таки открывает?

Да, в формате R11 открывает и читает, что уже очень радует и в принципе в инженерных задачах - более чем.
Можно закрывать трек. 

Хорошо бы добавить в wiki, чтобы они гуглились сюда, со временем.

XRay3D, 2021-12-02:
> Хорошо бы добавить в wiki, чтобы они гуглились сюда, со временем.

В новогодние возможно займусь справкой здесь.

closed 2021-12-02 by XRay3D