XRay3D/GERBER_X3 #58: редактирование формата сверловки

info69, 2021-08-19:
При редактировании формата сверловки не сохраняется новое значение
[image]

YuryW, 2021-08-22:
Это параметры для редактирования в текущем сеансе проекта .
А вот здесь они сохраняются.
[image]

info69, 2021-08-22:
Добрый день. В настройках нет вкладки excellon.
[image]

YuryW, 2021-08-22:
Следите за самыми свежими версиями среди тестовых. Это появилось с версии [0.8.0.2](https://disk.yandex.ru/d/L-laABnSVY4Ubw/0.8.0/0.8.0.2).
[image]

info69, 2021-08-22:
[image]

YuryW, 2021-08-22:
Это только часть программы.. Чтобы получить отдельно эту новую версию GGEasy, нужно сделать полную копию предыдущей и в ней заменить эти две папки и экзешник на новые:
[image]
[image]

На многие вопросы Вы найдёте ответы [тут](https://cnc-club.ru/forum/viewtopic.php?f=155&t=22277&view=unread#unread) и [тут](https://cnc3018.ru/threads/ggeasy-grber2gcode-frezerovka-pechatnyx-plat-na-chpu-zasvetka-lazerom-izuchaem-delimsja-opytom-i-dostizhenijami.2231/).

info69, 2021-08-23:
Спасибо

closed 2021-08-23 by XRay3D