XRay3D/GERBER_X3 #59: Зум колесиком мышки

vvv1954, 2021-08-28:
Нельзя ли сделать, чтобы зум менялся колесиком мышки? Как, например, в Aspire или VCarve.

XRay3D, 2021-09-06:
Можно 
Меняется здесь 
```cpp
void GraphicsView::wheelEvent(QWheelEvent* event)
{
    const auto delta = event->angleDelta().y();

#if (QT_VERSION < QT_VERSION_CHECK(5, 14, 0))
    const auto pos = event->pos();
#else
    const auto pos = event->position().toPoint();
#endif

    static auto sbUpdate = [&delta, this, scale = 3](QScrollBar* sb) { // Warning if create more GraphicsView`s!!
        if (App::settings().guiSmoothScSh())
            animate(sb, "value", sb->value(), sb->value() - sb->pageStep() / (delta > 0 ? +scale : -scale));
        else
            sb->setValue(sb->value() - delta);
    };

    if (event->buttons() & Qt::RightButton) {
        if (abs(delta) == 120) {
            setInteractive(false);
            if (delta > 0)
                zoomIn();
            else
                zoomOut();
            setInteractive(true);
        }
    } else {
        switch (event->modifiers()) {
        case Qt::ControlModifier:
            if (abs(delta) == 120) {
                setInteractive(false);
                if (delta > 0)
                    zoomIn();
                else
                    zoomOut();
                setInteractive(true);
            }
            break;
        case Qt::ShiftModifier:
            if (!event->angleDelta().x())
                sbUpdate(QAbstractScrollArea::horizontalScrollBar());
            break;
        case Qt::NoModifier:
            if (!event->angleDelta().x())
                sbUpdate(QAbstractScrollArea::verticalScrollBar());
            break;
        default:
            //QGraphicsView::wheelEvent(event);
            return;
        }
    }
    mouseMove(mapToScene(pos));
    event->accept();
    update();
}
```

vvv1954, 2021-09-07:
Спасибо. Какая версия VS?

XRay3D, 2021-09-08:
> Спасибо. Какая версия VS?

VS самая последняя и Qt5.12.11

XRay3D, 2021-12-02:
Будет переработано, рулетка отдельным инструментом станет.

XRay3D, 2024-03-08:
Колесо зумит при зажатой правой.

closed 2024-03-08 by XRay3D