XRay3D/GERBER_X3 #63: Compilation Error (Visual Studio)

stuartbarth, 2021-12-01:
Firstly, thank you for producing such great open-source software. This is probably the best user interface I have ever used for a program that generates GCode.

I have made some minor changes to the code outputs specific to my old milling machine, and would like to recompile the changes into a new binary. I am working in Visual Studio (2022), and I think have installed all the necessary dependencies.

The CMake runs fine, but when I try to build a release, I get the following error: `Error	D8021	invalid numeric argument '/Wno-error=invalid-offsetof'`. It may have something to do with clipper and shapes, but I'm not sure what.

The complete build output is shown below.

Regards!

> >------ Build All started: Project: GERBER_X3-0.8.0, Configuration: x64-Release ------
  [1/202] Automatic MOC and UIC for target shape
  [2/202] Automatic MOC and UIC for target clipper
  [3/202] Building CXX object shape_plugins\shape\CMakeFiles\shape.dir\shape.cpp.obj
  FAILED: shape_plugins/shape/CMakeFiles/shape.dir/shape.cpp.obj 
  C:\PROGRA~1\MICROS~4\2022\COMMUN~1\VC\Tools\MSVC\1430~1.307\bin\Hostx64\x64\cl.exe  /nologo /TP -DCTRE -DMINGW_HAS_SECURE_API=1 -DQT_CORE_LIB -DQT_GUI_LIB -DQT_MESSAGELOGCONTEXT -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DSHAPE_LIBRARY -DUNICODE -DWIN32 -DWIN64 -D_ENABLE_EXTENDED_ALIGNED_STORAGE -D_UNICODE -D_WIN32_WINNT=0x0601 -D_WIN64 -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\out\build\x64-Release\shape_plugins\shape -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\shape_plugins\shape -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\out\build\x64-Release\shape_plugins\shape\shape_autogen\include -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\magic_get -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\magic_get\include -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\compile-time-regular-expressions\single-header -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\shape_plugins\shape\..\..\GGEasy -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\shape_plugins\shape\..\..\static_libs\clipper -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\shape_plugins\shape\..\..\static_libs\common -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\shape_plugins\shape\..\..\static_libs\filetree -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\shape_plugins\shape\..\..\static_libs\gerber -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\shape_plugins\shape\..\..\static_libs\gi -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\shape_plugins\shape\..\..\static_libs\graphicsview -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\shape_plugins\shape\..\..\static_libs\thermal -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\shape_plugins\shape\..\..\static_libs\tooldatabase -IC:\Qt\6.2.1\mingw81_64\include\QtCore -IC:\Qt\6.2.1\mingw81_64\include -IC:\Qt\6.2.1\mingw81_64\mkspecs\win32-g++ -IC:\Qt\6.2.1\mingw81_64\include\QtGui -IC:\Qt\6.2.1\mingw81_64\include\QtWidgets /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MD /Zi /O2 /Ob1 /DNDEBUG -Wno-error=invalid-offsetof -std:c++20 /showIncludes /Foshape_plugins\shape\CMakeFiles\shape.dir\shape.cpp.obj /Fdshape_plugins\shape\CMakeFiles\shape.dir\shape.pdb /FS -c C:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\shape_plugins\shape\shape.cpp
C:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\out\build\x64-Release\cl : Command line error D8021: invalid numeric argument '/Wno-error=invalid-offsetof'
  [4/202] Building CXX object static_libs\clipper\CMakeFiles\clipper.dir\clipper_autogen\mocs_compilation.cpp.obj
  FAILED: static_libs/clipper/CMakeFiles/clipper.dir/clipper_autogen/mocs_compilation.cpp.obj 
  C:\PROGRA~1\MICROS~4\2022\COMMUN~1\VC\Tools\MSVC\1430~1.307\bin\Hostx64\x64\cl.exe  /nologo /TP -DCLIPPER_LIBRARY -DCTRE -DMINGW_HAS_SECURE_API=1 -DQT_CORE_LIB -DQT_GUI_LIB -DQT_MESSAGELOGCONTEXT -DQT_NO_DEBUG -DUNICODE -DWIN32 -DWIN64 -D_ENABLE_EXTENDED_ALIGNED_STORAGE -D_UNICODE -D_WIN32_WINNT=0x0601 -D_WIN64 -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\out\build\x64-Release\static_libs\clipper -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\static_libs\clipper -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\out\build\x64-Release\static_libs\clipper\clipper_autogen\include -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\magic_get -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\magic_get\include -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\compile-time-regular-expressions\single-header -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\static_libs\..\GGEasy -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\static_libs\..\static_libs\clipper -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\static_libs\..\static_libs\common -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\static_libs\..\static_libs\filetree -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\static_libs\..\static_libs\gerber -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\static_libs\..\static_libs\gi -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\static_libs\..\static_libs\graphicsview -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\static_libs\..\static_libs\thermal -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\static_libs\..\static_libs\tooldatabase -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\static_libs\..\shape_plugins\shape -IC:\Qt\6.2.1\mingw81_64\include\QtCore -IC:\Qt\6.2.1\mingw81_64\include -IC:\Qt\6.2.1\mingw81_64\mkspecs\win32-g++ -IC:\Qt\6.2.1\mingw81_64\include\QtGui /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MD /Zi /O2 /Ob1 /DNDEBUG -Wno-error=invalid-offsetof -std:c++20 /showIncludes /Fostatic_libs\clipper\CMakeFiles\clipper.dir\clipper_autogen\mocs_compilation.cpp.obj /Fdstatic_libs\clipper\CMakeFiles\clipper.dir\clipper.pdb /FS -c C:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\out\build\x64-Release\static_libs\clipper\clipper_autogen\mocs_compilation.cpp
C:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\out\build\x64-Release\cl : Command line error D8021: invalid numeric argument '/Wno-error=invalid-offsetof'
  [5/202] Building CXX object shape_plugins\shape\CMakeFiles\shape.dir\shape_autogen\mocs_compilation.cpp.obj
  FAILED: shape_plugins/shape/CMakeFiles/shape.dir/shape_autogen/mocs_compilation.cpp.obj 
  C:\PROGRA~1\MICROS~4\2022\COMMUN~1\VC\Tools\MSVC\1430~1.307\bin\Hostx64\x64\cl.exe  /nologo /TP -DCTRE -DMINGW_HAS_SECURE_API=1 -DQT_CORE_LIB -DQT_GUI_LIB -DQT_MESSAGELOGCONTEXT -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DSHAPE_LIBRARY -DUNICODE -DWIN32 -DWIN64 -D_ENABLE_EXTENDED_ALIGNED_STORAGE -D_UNICODE -D_WIN32_WINNT=0x0601 -D_WIN64 -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\out\build\x64-Release\shape_plugins\shape -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\shape_plugins\shape -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\out\build\x64-Release\shape_plugins\shape\shape_autogen\include -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\magic_get -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\magic_get\include -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\compile-time-regular-expressions\single-header -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\shape_plugins\shape\..\..\GGEasy -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\shape_plugins\shape\..\..\static_libs\clipper -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\shape_plugins\shape\..\..\static_libs\common -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\shape_plugins\shape\..\..\static_libs\filetree -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\shape_plugins\shape\..\..\static_libs\gerber -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\shape_plugins\shape\..\..\static_libs\gi -IC:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\shape_plugins\shape\..\..\static_libs\graphicsview -IC:\Users\sbart\Documents\Projects\KLG\GERBER_ [... cut ...] pdb /FS -c C:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\static_libs\clipper\myclipper.cpp
C:\Users\sbart\Documents\Projects\KLG\GERBER_X3-0.8.0.3_Modified\out\build\x64-Release\cl : Command line error D8021: invalid numeric argument '/Wno-error=invalid-offsetof'
  [10/202] Automatic MOC and UIC for target gi
  [11/202] Automatic MOC and UIC for target thermal
  [12/202] Automatic MOC and UIC for target tooldatabase
  ninja: build stopped: subcommand failed.

Build All failed.

XRay3D, 2021-12-06:
You're compiler support c++20 standard?

stuartbarth, 2021-12-07:
Thanks for the tips.
For some reason, my Qt installation did not install all the required MSVC files.
Reinstalling, specifying Qt5 and removing CMAKE references to Qt 6 solved the issue and compilation completed successfully.

Thank you!

closed 2021-12-07 by stuartbarth