XRay3D/GERBER_X3 #72: HLDI. Включения лазера при движении в одном направлении по оси.

carrotFoxx, 2022-02-06:
Добрый день Дамир и прочие участники проекта.
Спасибо за вашу работу. 

Есть ли возможность добавить функцию при использовании режима HLDI, позволяющая активировать лазер при движении только в одном направлении по оси?
Это позволит снизитьпогрешность, вызванную люфтами в механизме станка.

XRay3D, 2022-08-05:
Типа развёртки телевизора?

sleeply4cat, 2025-07-12:
ага, вроде того. Иначе гребёнка получается на краях

[image]

zhusmansan, 2026-01-20:
> ага, вроде того. Иначе гребёнка получается на краях
> 
> <img alt="Image" width="1280" height="960" src="https://private-user-images.githubusercontent.com/37955805/465612047-ed7a8ea6-5383-4940-85e2-d2e9b32ebd39.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Njg5MjYwNTEsIm5iZiI6MTc2ODkyNTc1MSwicGF0aCI6Ii8zNzk1NTgwNS80NjU2MTIwNDctZWQ3YThlYTYtNTM4My00OTQwLTg1ZTItZDJlOWIzMmViZDM5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNjAxMjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjYwMTIwVDE2MTU1MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTViNTIyYzEzMmI4MDIxNjg4MTM2MGU1NGQwNzA4MjQwYTcwZjI3M2MxZTQyOThjZGI2ZDcxZWM2NjVmMGE3YzkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.pjpdScijG_384LPergBcZaCwc_aBRIqQ8NfaZClqIpw">

Писал ровно такую фичу ровно для тех-же целей. Правда в другой софтине. см https://github.com/johannesnoordanus/image2gcode/tree/main