XRay3D/GERBER_X3 #73: Возможность отрицательного отступа

carrotFoxx, 2022-02-06:
Добрый день, коллеги!

Есть ли возможность сделать допустимым установку отступа равной 0 или отрицательному значению, т.к. в случае с HLDI возможна "недозаливка" (см. скриншоты)

[image]
 
[image]

YuryW, 2022-02-13:
Добрый вечер!
Отступ (смещение) - это расстояние между двумя центрами соседних проходов лазера или фрезы. 10% отступа позволяет делать перекрытие площади освещения (фрезерования)  на 90% между двумя соседними проходами луча (фрезы). И это сейчас возможно выставить в HLDI. При нулевом отступе инструмент будет перемещаться бесконечно на одном месте. При отрицательном - изменит направление последовательного смещения на противоположное только лишь.
От края меди смещение не задается пользователем и программно равно всегда половине диаметра инструмента.

XRay3D, 2022-09-25:
Если так боитесь, то можете добить профилем поверху или снаружи.

closed 2022-09-25 by XRay3D