XRay3D/GERBER_X3 #75: Сохранение УП c кодировкой UTF-8

RomSunZ, 2022-02-12:
Сейчас при сохранении приходится открывать УП в Notepad++ и руками менять кодировку на UTF-8. В противном случае при работе с LinuxCNC УП не отображается в окне просмотра кода из-за неверных символов переноса строки. Было-бы более правильно при сохранении УП сразу указывать кодировку как UTF-8.

ronindev, 2023-05-19:
Символы переноса строки не зависят от кодировки же. Если вы создаёте УП по Windows или Mac, то возможно будут проблемы с переносом. Я делаю под Linux и исполняю тоже под Linux - никаких проблем

RomSunZ, 2023-05-19:
Я наверное не верно написал, разные символы переноса в windows/linux для кодировок отличных от UTF. Я создаю УП в windows и потом переношу на Linux для работы в LinuxCNC. И для работы УП на линуксе приходится вручную менять кодировку УП на utf.

ronindev, 2023-05-19:
Кажется это должно помочь вам:
`Index: plugins/gcode/gc_file.cpp
IDEA additional info:
Subsystem: com.intellij.openapi.diff.impl.patch.CharsetEP
<+>UTF-8
===================================================================
diff --git a/plugins/gcode/gc_file.cpp b/plugins/gcode/gc_file.cpp
--- a/plugins/gcode/gc_file.cpp	(revision 9204db96e8820b73b6b99ded60931c43526a95bf)
+++ b/plugins/gcode/gc_file.cpp	(date 1681547393632)
@@ -138,7 +138,7 @@
             if (!s.isEmpty())
                 str.push_back(s);
             if (!str.endsWith('\n'))
-                str.push_back("\n");
+                str.push_back("\r\n");
         }
         out << str;
     } else
`

ronindev, 2023-05-19:
Кажется это должно помочь вам:

```
Index: plugins/gcode/gc_file.cpp
IDEA additional info:
Subsystem: com.intellij.openapi.diff.impl.patch.CharsetEP
<+>UTF-8
===================================================================
diff --git a/plugins/gcode/gc_file.cpp b/plugins/gcode/gc_file.cpp
--- a/plugins/gcode/gc_file.cpp	(revision 9204db96e8820b73b6b99ded60931c43526a95bf)
+++ b/plugins/gcode/gc_file.cpp	(date 1681547393632)
@@ -138,7 +138,7 @@
             if (!s.isEmpty())
                 str.push_back(s);
             if (!str.endsWith('\n'))
-                str.push_back("\n");
+                str.push_back("\r\n");
         }
         out << str;
     } else
```