XRay3D/GERBER_X3 #78: Внутренний отступ от краёв дорожек/падов.

beerurri, 2022-02-23:
Хотелось бы иметь фичу, которая позволяла бы настраивать отступ от краёв дорожек/падов в мм или проходах.
Засвечиваю фоторезист лазером ("Карман"), и, если засвечивать прямо до краёв падов/дорожек, из-за неидеальности и люфтов конструкции засвечивается лишнее, и близко расположенные компоненты, в конце концов, становятся единым целым.
Например, если измерять эту фичу в проходах инструмента (лазера), то я бы убрал крайний проход (зачёркнутый на скриншоте)

[image]

Const55, 2022-02-24:
Поддерживаю просьбу !
также при гравировке печатных плат фрезой хотелось бы иметь такой функционал 
пример: первый проход УП профиль Делаем тонкой Фрезой 0,1мм 20 градусов 
Второй проход УП Карман более крупной Фрезой 0,3мм 60 градусов вычищаем медь за 2-3 круга  вот тут и надо отступить 0,1мм от контура дорожки и не зацепить ее край,  нужно учесть что граверы с канавками имеют биение на конце как минимум 0,1мм

sslobodyan, 2022-03-20:
Укажите диаметр вашего инструмента бОльшим, чем реальный диаметр, и получите отступ от краев. Только не забудьте перекрытие проходов увеличить.

Const55, 2022-03-21:
> Укажите диаметр вашего инструмента бОльшим, чем реальный диаметр, и получите отступ от краев. Только не забудьте перекрытие проходов увеличить. Например, фреза реально режет 0.2мм а вы указываете в параметрах фрезы 0.3мм. Получите отступ в 0.05мм (0.3-0.2)/2 .
Спасибо за совет попробую таким способом,  то есть  исходя из вашего примера Для указанной Фрезы 0,3 мм  стандартный отступ 50%  = 0,15мм   указываем  0,15 +  (0.3-0.2)/2 = 0,2  так ??

XRay3D, 2022-11-16:
> Спасибо за совет попробую таким способом,  то есть  исходя из вашего примера Для указанной Фрезы 0,3 мм  стандартный отступ 50%  = 0,15мм   указываем  0,15 +  (0.3-0.2)/2 = 0,2  так ??

Лазер = 0.2 + люфты ~~ 0.3 но перекрытие указываем как для 0.2 мм, то есть 0.1 мм (50%).

XRay3D, 2023-05-17:
> Всё равно хотелось бы это отдельной опцией)

Ладно, когда припуск на чистовой проход добавлю его опционально отдельно сохранять буду.