XRay3D/GERBER_X3 #82: Изображение улетает за границы viewport

pushkin42, 2022-05-25:
**Versions**
 - GGEasy version: 0.8.0.1 и выше (пробовал обе)

**PC info**
- OS: Windows Server 2016 R2 x64
- CPU: Intel Xeon E5450v2
- RAM: 64Gb
- DirectX 12 at NVidia GeForce 1050Ti
- Разрешение экрана 1366x768

**Describe the bug**
При открытии любого гербера вьюпорт масштабируется до невероятных размеров, помогает только принудительное выставление масштаба 100%, тогда изображение появляется.

**To Reproduce**
1. Откройте программу
2. Откройте любой набор герберов
3. Показывает черный экран :)
4. ????
5. PROFIT!

**Expected behavior**
Хотелось бы, чтобы герберы открывались в рамках вьюпорта целиком.
Хотя, если программу не закрывать и открыть новый набор файлов, глюк не наблюдается...

**Screenshots**
вот так это должно быть (после ручного масштабирования):
[image]

.

pushkin42, 2022-05-25:
[image]
вот обнаружил полосочку еще какую-то. в файлах ее нету.

XRay3D, 2022-08-05:
А если без сверловок?

closed 2025-05-12 by XRay3D