XRay3D/GERBER_X3 #86: Форк проекта

GogoLike, 2022-09-29:
Здравствуйте, при открытии вашего проекта в QtCreator на разных версиях _Qt_ (_5.15.2, 6.3.0, 6.4.0_) не удается найти большое число файлов. Можно ли их добавить в репозиторий? Например папка _/GGEasy/forms/_
Хочу самостоятельно собрать ваш проект и добавить некоторые команды для своей работы.

kostyabelov, 2022-09-29:
Сегодня собрал, чтобы пофиксать самостоятельно баг "Штриховка не порождает УП #69", с которым сам столкнулся, ну и победил :-)
1. Встаньте на tag: 0.8.1.0, это последний релиз.
2. Для компиляции нужен как QT6, так и QT5, устанавливается все через online installer.
3. Конфиг и сборка cmake, для компиляции я использовал gcc (mingw).
4. compile-time-regular-expressions, сюда нужно вытянуть репозитарий с одноименным названием.
5. Я не использовал qt creator, все ручками. vim рулит :-)
6. Сходу не удалось статически прилинковать libstdc++, по-этому пришлось dllки приложит в папку запуска.

kostyabelov, 2022-09-29:
2.1. В Qt6: нужен serialport, в Qt5: script.
3.1. + ninja

XRay3D, 2022-11-13:
> 2.1. В Qt6: нужен serialport, в Qt5: script. 3.1. + ninja

??? странно.

> Здравствуйте, при открытии вашего проекта в QtCreator на разных версиях _Qt_ (_5.15.2, 6.3.0, 6.4.0_) не удается найти большое число файлов. Можно ли их добавить в репозиторий? Например папка _/GGEasy/forms/_ Хочу самостоятельно собрать ваш проект и добавить некоторые команды для своей работы.

Вы рекурсивно стянули? Там ещё зависимости с других реп есть.