XRay3D/GERBER_X3 #89: При экспорте из DipTrace файла сверловки изображение сверловки становится больших размеров. Ошибка.

Pank172, 2022-10-31:
Versions

GGEasy version: 0.8.1.0 
PC info

OS: Windows 10 x64
CPU: AMD Ryzen 3600
RAM: 16Gb
DirectX 12 at Radeon RX570
Разрешение экрана 1920х1080

Использую программу Diptrace 4.3.0.1.
При переносе в программу  файла сверловки из Diptrace размер отверстий стает больше размера платы в 50 раз.

XRay3D, 2022-11-03:
Можете прикрепить файл или привести текст файла сверловок.

Pank172, 2022-11-03:
Да конечно
[444-1-2_gerber.zip](https://github.com/XRay3D/GERBER_X3/files/9928266/444-1-2_gerber.zip)

Pank172, 2022-11-04:
Размер сверловки в 10 раз превышает размеры платы. Размер платы 95Х105 мм, а размер сверловки 950Х1050мм. Как побороть?

Pank172, 2022-11-04:
Разобрался. Установил соотношение 1:2 в настройках файла сверловки. Единицы - миллиметры, нули - ведущие, формат - 1:2, смещение Х-0,0000, У - 0,0000. коэффициент - х10.

Pank172, 2022-11-04:
Спасибо за классную программу.

closed 2022-11-04 by Pank172