XRay3D/GERBER_X3 #91: Немного переводов и мелких правок

GeorgKZ, 2023-03-16:
Это для ветки dev_clipper_2 недельной давности. Убирал предупреждения при сборке, некоторые предупреждения при работе, доделывал русификацию. Собирал под Qt-6.4.2 под Линукс.

Ещё в принципе есть наработки статической сборки под Ликус и Windows, но для Qt с 6й версии.

Патч положил тут:
[https://drive.google.com/file/d/1XCASbYDJGWnU7nu364RmKggseki_C2Wx/view?usp=sharing](https://drive.google.com/file/d/1XCASbYDJGWnU7nu364RmKggseki_C2Wx/view?usp=sharing)

closed 2023-03-24 by XRay3D

GeorgKZ, 2023-03-24:
Я извиняюсь, поторопился с выкладиванием. Вот тут архив:
https://drive.google.com/file/d/1XjHG_CIRC9Hl6Hk7TygCrw8mUmjbHwDp/view?usp=sharing
почистил, разобрал и подписал что для чего нужно. Для справки вот
https://drive.google.com/file/d/1OCGojDlqDImpaBTNL6Pmd_dKEAXLpjJZ/view?usp=sharing
вариант с применёнными патчами, чтобы было понятно, какие файлы обработаны. Исходный архив, к которому были применены патчи:
https://drive.google.com/file/d/1OCysd5PdZ1tZ9mZcKzWLzUuUePFHemsU/view?usp=sharing

GeorgKZ, 2023-03-24:
Собирал под Линукс примерно так (в /usr/local/Qt-6.4.2-shared/ был собран Qt):

ulimit -n 2048

CMAKE_PREFIX_PATH="/usr/local/Qt-6.4.2-shared/lib/cmake"
PATH="/usr/local/Qt-6.4.2-shared/bin:$PATH"
BUILDDIR="../GGEASY-build-6.4.02-shared-deb"
SRCDIR="../GGEASY"

# Конфигурация
cmake -G Ninja "-DCMAKE_EXE_LINKER_FLAGS=-pthread" -DCMAKE_BUILD_TYPE=Debug   -S ${SRCDIR} -B ${BUILDDIR}

# Сборка
cmake --build ${BUILDDIR} --target update_translations --target all --parallel > build_6.4.02_shared_deb_log.txt

# Добавление переводов Qt
cp -fs /usr/local/Qt-6.4.2-shared/translations/qtbase_ru.qm ../GGEASY/bin/Qt6.4_gnu_d_x64/translations/
cp -fs /usr/local/Qt-6.4.2-shared/translations/qtbase_en.qm ../GGEASY/bin/Qt6.4_gnu_d_x64/translations/

XRay3D, 2023-03-25:
Почему через гитхаб не захотели мне изменения залить?
Гляну. Чищу код пока, многое не нравится.

GeorgKZ, 2023-03-25:
Я пока с гитхабом как-то не общался, только начал пробовать.

Если по моим предложениям будут вопросы - пишите, постараюсь подробно ответить.

Кодинг нормальный, только непривычно много новшеств C++. Я пока только 11-е нововведения смотрю, а уже 20й вовсю используется. А ведь только недавно Фортран на перфокартах был...

И по личному мнению, неплохо было бы доксигенности добавить - ну это в основном для облегчения другим разбираться, что есть что. Для примера комментирования вывалил исходники неспешно ваяемой программки (https://github.com/GeorgKZ/stl2step) - вот типа такого, со ссылками на ISO 06983-1-2009 и т.п.

GeorgKZ, 2023-03-28:
Обнаружил у себя ошибку: в файле voronoi_boost.cpp по хорошему надо было так:
`    #if defined(_MSC_VER)
        #pragma warning(push)
        #pragma warning(disable : 5055)
    #elif defined(__GNUC__) && (__GNUC__ >= 7)
        #pragma GCC diagnostic push
        #pragma GCC diagnostic ignored "-Wpragmas"
        #pragma GCC diagnostic ignored "-Wdeprecated-enum-float-conversion"
    #endif /* _MSC_VER, __GNUC__ */

    #include "voronoi_visual_utils.h"
    #include <boost/polygon/polygon.hpp>
    #include <boost/polygon/voronoi.hpp>

    #if defined(_MSC_VER)
        #pragma warning(pop)
    #elif defined(__GNUC__) && (__GNUC__ >= 7)
        #pragma GCC diagnostic pop
    #endif /* _MSC_VER, __GNUC__ */
`
а то в том варианте при сборке MS VCPP этот блок пропускался, выходила ошибка сборки.

И ещё про сборку для Windows. Вот в этом фрагменте CMakeLists.txt для исполняемого файла:
`set_target_properties(${PROJECT_NAME} PROPERTIES
    MACOSX_BUNDLE_GUI_IDENTIFIER xr.ggeasy.com
    MACOSX_BUNDLE_BUNDLE_VERSION ${PROJECT_VERSION}
    MACOSX_BUNDLE_SHORT_VERSION_STRING ${PROJECT_VERSION_MAJOR}.${PROJECT_VERSION_MINOR}
    MACOSX_BUNDLE TRUE
    WIN32_EXECUTABLE TRUE
)`
Если закомментировать или убрать WIN32_EXECUTABLE TRUE (то есть в опциях линкера заменить /subsystem:windows на /subsystem:console), то восстанавливается вывод в консоль - все предупреждения, qDebug и т.д. Возможно, стоит ввести отдельный флаг сборки (для переключения этого параметра) для удобства отладки. Ну или отметить комментарием такую возможность.
На линукс-сборку это не влияет - там вывод в консоль идёт в любом случае.

GeorgKZ, 2023-04-04:
Не только. Я имел в виду статическую сборку Вашей программы - для этого необходимо конечно использовать статически собранный вариант Qt, а в Вашей программе все плюгины и все библиотеки Qt будут интегрированы в единый исполняемый файл. Правда, "универсальный" (то есть, если собирать динамически с динамическим Qt, то получится как раньше, а если статически со статически собранным Qt - то единым исполняемым файлом) вариант, насколько я понял, получится только с Qt с 6й версии - с использованием для плюгинов вместо 
`add_library(rectangle SHARED
        rectangle.json
        shrectangle.cpp
        shrectangle.h
        )`
конструкции
`    qt_add_plugin(rectangle CLASS_NAME PluginRect)
    target_sources(rectangle PRIVATE
        rectangle.json
        shrectangle.cpp
        shrectangle.h
        )
`
где PluginRect - это бывший class Plugin, и в основной программе для подключения плюгинов прямое обращение к объектам вместо файлов:
`#else // STATIC_LINK
        QObject* objects[] = {
                              new Gerber::PluginGerber(),
                              new Dxf::PluginDxf(),
                              new Excellon::PluginExcellon(),
                              new Hpgl::PluginHpgl(),
                              new Shapes::PluginCircle(),
                              new Shapes::PluginArc(),
                              new Shapes::PluginRect(),
                              new Shapes::PluginText(),
                              new Shapes::PluginPoly(),
                              nullptr}; 

        for (auto s = &objects[0]; *s != nullptr; ++ s) {

            QObject* pobj = *s;
            if (auto file = qobject_cast<FilePluginInterface*>(pobj); pobj && file) {
                App::filePlugins().emplace(file->type(), PIF { file, pobj });
                continue;
            }
            if (auto shape = qobject_cast<ShapePluginInterface*>(pobj); pobj && shape) {
                App::shapePlugins().emplace(shape->type(), PIS { shape, pobj });
                continue;
            }
        }
#endif
`

XRay3D, 2023-04-06:
Когда нибудь)) когда АПИ устаканится, я на оборот монолит распилил что бы в будущем добавлять функциональности проще было. Сейчас репу чешу в сторону https://gitlab.com/inkscape/lib2geom
Но её что то собрать не очень выходит.