XRay3D/GERBER_X3 #98: Сборка и ctre

shprotru, 2023-05-25:
Попытался собрать в линукс `Ubuntu 22.04.2 LTS`, собирал cmake'ом следующим образом:
`mkdir build && cd build && cmake -DCMAKE_BUILD_TYPE=Release ..`
столкнулся с тем что GERBER_X3 в некоторых своих плагинах использует `ctre`, ранее я никогда не сталкивался с такой зависимостью. Поиск в штатных репозиториях `apt` не увернчался успехом. Нагуглил возможность установки при помощи [vcpkg](https://github.com/hanickadot/compile-time-regular-expressions), выполнил всё по инструкции, программа по прежнему не собиралась, поскольку не видела зависимость. Тогда я воспользовался рекомендацией самого vcpkg и полез дописывать её в модули которые используют эту зависимость. Их там действительно не оказалось, что несколько странно, после внесения этих изменений сборка прошла успешно.
Прилагаю патч, или заводить запрос внесения изменений?
[ctre.patch.txt](https://github.com/XRay3D/GERBER_X3/files/11570048/ctre.patch.txt)

XRay3D, 2023-06-19:
git submodule update --init --recursive или что-то подобное попробуйте. Она с гита погрузиться в external_dependencies.

ASRSim13, 2023-07-14:
не подгрузится. Ругаться на права начинает.Но можно зайти в external_dependencies и от туда git clone https://github.com/hanickadot/compile-time-regular-expressions.git
И вторую внешнюю зависимость так же.

ASRSim13, 2023-07-31:
При доступе через ssh - загружает внешние зависимости, только придётся зарегистрироваться на github и добавить ssh ключь в настройках.

mikelsv, 2024-02-09:
GitHub Desktop загружает основной проект, дальше ругается на ssh и выходит с ошибкой. Скачать отдельно через git clone помогло. Где почитать, как настроить ssh под windows?

Какая минимальная версия Qt необходима? Смог скачать Qt5.14.2, её и поставил. Получаю различные ошибки. error C2039: "toDouble": не является членом "QStringView".  C2665 "QObject::connect": ни одна перегруженная функция не может преобразовать все типы аргументов.
Собираю в 2022 студии. В Cmake указал Qt/msvc2017_64

Собственно, GGEasy вылетает после сохранения gcode, если долго сохранять в один и тот же файл. Хочу запустить с отладчиком, понять причину. Ну и другие проблемы попробовать исправить.

kostyabelov, 2024-03-09:
.git/config:

[submodule "external_dependencies/compile-time-regular-expressions"]
    active = true
    url = https://github.com/hanickadot/compile-time-regular-expressions.git
[submodule "external_dependencies/pfr_non_boost"]
    active = true
    url = https://github.com/apolukhin/pfr_non_boost.git

XRay3D, 2024-03-10:
Лучше выкачать рекурсивно ```dev900b``` и закомментить строчки в Clipper2 симэйклисте.
```
# option(CLIPPER2_UTILS "Build utilities" ON)
# option(CLIPPER2_EXAMPLES "Build examples" ON)
# option(CLIPPER2_TESTS "Build tests" ON)
# option(USE_EXTERNAL_GTEST "Use system-wide installed GoogleTest" OFF)
# option(USE_EXTERNAL_GBENCHMARK "Use the googlebenchmark" OFF)
# option(BUILD_SHARED_LIBS "Build shared libs" OFF)
```
и добавить clipper.engine.h в строке 363
```cpp
auto begin() { return childs_.begin(); }
auto end() { return childs_.end(); }
```

XRay3D, 2024-07-20:
> ```cmake
> option(CLIPPER2_UTILS "Build utilities" ON)
> ...
> option(BUILD_SHARED_LIBS "Build shared libs" OFF)
> ```
> 
> и добавить clipper.engine.h в строке 363

вот я клубень в СМаке !

надо было перед включением 
```cmake
set(CLIPPER2_UTILS OFF)
set(CLIPPER2_EXAMPLES OFF)
set(CLIPPER2_TESTS OFF)
```
добавить и всё

closed 2025-05-12 by XRay3D