XRay3D/GERBER_X3 #101: Артефакты с контактными площадками SMD компонентов, установленных не под прямым углом

DannyColdburn, 2023-09-19:
При загрузке гербер файла возникают артефакты с контактными площадками SMD компонентов, которые установлены не под прямым углом.
Проблема касается контактных площадок с формой Rounded Rectangle и Oval.
[image]

XRay3D, 2024-02-09:
Файл приложите. А то оно 100500 вариантов может использовать для форммирования этих прощадок.

Maximych, 2024-03-08:
А что с этими площадками?
[Gerber_TL-kontroler_PCB_TL-kontroler_2_2024-03-08.zip](https://github.com/XRay3D/GERBER_X3/files/14537078/Gerber_TL-kontroler_PCB_TL-kontroler_2_2024-03-08.zip)

XRay3D, 2024-03-08:
Добавлю в копилку тестов. Макрос неправильно разбирается.

commit 0316012a (2024-03-08): close #101; Fix ApMacro::drawVectorLine
closed 2024-03-08 by XRay3D