XRay3D/GERBER_X3 #110: Проблемы и предложения

mikelsv, 2024-02-11:
Список проблем и предложений, который возникли в процессе использования GGEasy.

1. Версия 0.8.1.0. Открыть проект https://drive.google.com/file/d/1dKDmDB8rIrct-dYG9j6LD_u3nrAUZlyZ/view?usp=sharing . Создать любой УП. Например, для сверления. Сохранить как Drill.tap . Программа сохраняет новый файл и вылетает с ошибкой. В собранной 0.8.6.0 эту ошибку пока повторить не удалось.

---
Новая 0.8.0.6 собранная из исходников.

2. Создаю новый проект. Добавляю все гербер файлы. Получаю несколько пустых окошек. В чём их смысл?
[image]

3. В программе английские надписи, хотя в настройках стоит русский язык.

4. Не понимаю логику названий. Close - удаляет из списка один файл. Delete Selected выбранные. Предлагаю Close Selected. Иначе возникает подозрение, что будут удалены реальные файлы.
[image]

5. G-Code Zero Point. Хотелось бы опцию автоматического перемещения в угол платы. А так же перемещение в эту же точку начала координат. И да, наибольшая проблема, координаты в нижнем трее показываются только от начала координат. Очень не хватает координат от G-Code. Там куча места. Autolpace All Refpoints не работает. Точки уходят в левый верхний угол. 

6. Сверловка.
6.1 - Галочка перед Aperture / Tool, работает только если выбраны инструменты для всех размеров отверстий. Если один инструмент не выбран, то нажатие на неё ничего не даёт и приходится ставить галочки вручную. По логике она должна включать / отключать доступные.
6.2 Не хватает ещё одной строчки Selected. Сверлить выбранные отверстия. Иначе приходится устраивать танцы с бубном и удалять лишние элементы из схемы. Мне пришлось делать много лишних действий.

7. Вылет программы. 0.8.6.0. Проект: https://drive.google.com/file/d/1Th8ZQ1Di-HBkTN2gYVXJoLxMCrpI_Y3h/view?usp=sharing . Выбрать контуры трансформатора T1. Обработка Pocket Raster. Inside. Создать. Программа вылетает.
Ошибка в File::createGiRaster(). for (size_t j = 0; j < paths.size() - 1; ++j) при пустом paths цикл улетает в минус бесконечность.

8. Колёсико мышки смещает схему по вертикали. Эта функция не имеет смысла. Предлагаю сделать как в KiCad. Управлять колёсиком изменением масштаба. Хотя уже привык наживать Ctrl + колёсико мышки. Но зачем нажимать лишнюю кнопку. Как вариант, сделать включение этой опции в настройках.

9. Кнопка Resize после Ruller прячет окно куда-то за пределы экрана. Если вытащить обратно и развернуть на весь экран, то окно снова улетает за пределы. После перезапуска снова работает нормально.

10. Нужна опция штриховки фигур. Иначе на месте конденсаторов, да и прочего, приходится рисовать круги. Чтобы сделать углубления. Так же не работает кнопка Escape при создании круга. Не туда нажал, как отменить?
Сделать углубление для конденсатора. Сейчас это боль. Я привык к OpenSCAD, где всё измеряется в точных цифрах. Работа мышкой смотрится довольно криво. Попасть в центр круга. Сделать правильный радиус. Есть трансформация, но там нет ни координат, ни радиуса. 
[image]

В общем, тут хватает проблем. И того, что хотелось бы улучшить.
Сейчас передохну и попробую что-то из этого улучшить, на что хватит понимания и накидать пулл реквестов.

mikelsv, 2024-02-14:
Отправил три пулл реквеста для тех улучшений, что были мне по силам. Хотелось бы понять дальнейший вектор приложения усилий.
1. Улучшения принимаются. В этом случае я буду разбираться с остальным списком проблем и вносить новые изменения.
2. Улучшения требуется доработать. Здесь нужна информация о том, что исправлять.
3. Улучшения не нужны. Тут я буду развивать свой форк с тем функционалом, который хочу использовать.

Пока у меня есть возможность и мотивация. Хотелось бы использовать её эффективно и вложить в улучшение хорошего проекта.
Так же хотелось бы иметь возможность получать обратную связь и ответы на вопросы по структуре.

kostyabelov, 2024-02-23:
Штриховка.

На любом куске текстолита 10см на 10см найдется 1см на 1см "ямы", из-за которой приходится углублять гравер и перештриховывать всю плату. 

ОЧЕНЬ хотелось бы иметь возможность ВЫДЕЛИТЬ часть уже готовых "проходок" созданной УП и пересобрать ее в частичную УП, куда попадают только те проходы которые "задевают" выбранный прямоугольник.

XRay3D, 2024-03-07:
Лучше дождаться 0.9.0.0 и уже там добавлять хотелки. Осталось немного. Примерно месяц. 0.8.х.х монолит неудобный для расширения.

ronindev, 2024-07-14:
Есть новости по 0.9.0.0?

XRay3D, 2024-07-20:
> Есть новости по 0.9.0.0?
Мало по малу движется. Пока ошибки и очевидные косяки лечу.

ronindev, 2024-10-08:
Очень хочется наконец дождаться 0.9.0.0 версии, чтобы больше людей подключились к разработке. Мне было бы достаточно чтобы эта версия собиралась под linux.
Выше человек тоже готов был присоединиться к разработке

ronindev, 2024-10-09:
> Штриховка.
>
> На любом куске текстолита 10см на 10см найдется 1см на 1см "ямы", из-за которой приходится углублять гравер и перештриховывать всю плату.
>
> ОЧЕНЬ хотелось бы иметь возможность ВЫДЕЛИТЬ [...]

это должно решаться на уровне снятия карты высот, кмк

XRay3D, 2024-10-13:
Coming soon))

mikhailmihalkov, 2024-12-16:
Вопрос можно ли после редактирования сохранить в формате гербер ?

XRay3D, 2024-12-30:
> Вопрос можно ли после редактирования сохранить в формате гербер ?

Только чтение.

ronindev, 2025-02-06:
Как там продвигается?)

ronindev, 2025-05-05:
[image]

XRay3D, 2026-04-08:
> 1. Опция Pocket Raster HLDI работает криво. Посылает лазер по оси Х за пределы контура платы от -2223,964 до 2423,19. И, как следствие, приводит к ошибке "выхода за пределы рабочей зоны".
> 2. Не все кнопки меню переведены на русский. А в v0.8.6.4 не переведены даже те, что в v0.8.1.0 были переведены, как это поправить?
> 3. Разные версии программ не открывают файлы *.g2g друг-друга. Даже более поздние версии не открывают файлы, созданные ранее. Как сохранять проекты при апдейте?

Выставьте правильную скорость лазера и ускорение, соответствующие вашему станку, перед созданием УП.
https://cnc3018.ru/threads/nastrojki-lazera-v-ggeasy-g2g.9890/post-226596

4ygeca, 2026-10-09:
0.9 версия хорша, добавление постпроцессора для создания УП отличный ход, теперь можно много хотелок добавить, просто редактируя js файлы. Я уже добавил себе подвод с одной стороный по осям при сверловке и пекинг для 0.3 сверла. Спасибо за работу.
Единственное, что нельзя там прикрутить и что действительно нужно добавить в афинные преобразования сдвиг по осям, т.е. из прямоугольника получить параллелограмм. Для оси Хэто будет как: x`=x+y*shx; y`=y, для Y: x`=x; y`=x*shy+y.
Потому как станки у всех не идеальные и перекос фрейма относительно стола есть у всех с той или иной степенью точности, а компенсировать это негде по сути, чтобы получить по итогу прямоугольник.
Когда все операции по созданию ПП делаются на одном станке это не критично и не заметно, одна и та же неточность повторяется из операции к операции, но если использовать фотошаблоны или светить на другом лазерном станке, где свои неточности, получается каша, часть отверстий попадает, часть уплывает.
Еще одной из хотелок это компенсация подтравливания. Понимаю, что при фрезеровке, а это основная задача программы это бессмысленная опция и фреза\грвер и так уже отводится при погружении, но при засветке фоторезиста лазером и последующем травлении, это достаточно нужная и не очевидная вещь.

XRay3D, 2026-11-01:
> 15\. 0.9 версия хорша, добавление постпроцессора для создания УП отличный ход, теперь можно много хотелок добавить, просто редактируя js файлы. Я уже добавил себе подвод с одной стороный по осям при сверловке и пекинг для 0.3 сверла. Спасибо за работу.
>      Единственное, что нельзя там прикрутить и что действительно нужно добавить в афинные преобразования сдвиг по осям, т.е. из прямоугольника получить параллелограмм. Для оси Хэто будет как: x`=x+y*shx; y`=y, для Y: x`=x; y`=x*shy+y.

Теоретически можно но тогда G2/3 придётся как то аппроксимировать и пересчитывать.