XRay3D/GERBER_X3 #120: Режим HLDI

inscrut, 2024-03-28:
Подскажите, может я не понимаю как верно пользоваться этой штриховкой:
инструмент лазер, выделяю контур (юзаю .dxf, но вроде и фигуры также работают), генерирую - все есть, треки есть, но почему то "холостые перемещения" выходят за границы поля и экрана в целом, иными словами - как ограничить, чтобы лазерная головка бегала внутри квадрата 50х50 мм условно, гравируя по центру кружок. Я выбирал по разному - везде гравируется кружок, но холостые перемещения - 500 мм влево, 500 мм вправо, условно. А у меня поле 300х180 мм :)

Если некорректно объяснил, снизу скрин
[image]

redbobcat, 2024-04-01:
Попробуйте поставить ускорение побольше, у меня дефолтом стояло 1mm/s2 и тоже вылет был очень большой. Поставил на свои реальные 200 и стало реалистично, чтобы каретка успела и тормознуть и снова разогнаться.

inscrut, 2024-04-01:
> Попробуйте поставить ускорение побольше, у меня дефолтом стояло 1mm/s2 и тоже вылет был очень большой. Поставил на свои реальные 200 и стало реалистично, чтобы каретка успела и тормознуть и снова разогнаться.

Спасибо, это звучит логично, просто не обратил внимание на этот параметр, попробую на днях. Думаю вопрос будет полезен через поисковик.

closed 2024-04-01 by inscrut